ToyKeeper/anduril #13: ./make anduril - No such file or directory

GarikFirst, 2023-11-07:
I’m running GNU bash, version 5.2.15(1)-release (aarch64-apple-darwin22.1.0) from homebrew on macOS Sonoma 14.1, globstar option is on:
```
MacBook-Pro14-username:anduril username$ shopt globstar
globstar       	on
```

But all builds failing with **invalid shell option name** and of course **No such file or directory**

```
MacBook-Pro14-username:anduril username $ ./make anduril
./make: line 11: shopt: globstar: invalid shell option name
===== anduril : fireflies-e01- =====
ui/anduril/anduril.c:45:23: fatal error: hw/fireflies/e01: No such file or directory
   45 | #include incfile(CFG_H)
      |                       ^
compilation terminated.
ERROR: build failed
===== anduril : fireflies-pl47--219 =====
ui/anduril/anduril.c:45:23: fatal error: hw/fireflies/pl47: No such file or directory
   45 | #include incfile(CFG_H)
      |                       ^
compilation terminated.
ERROR: build failed
```

SiteRelEnby, 2023-11-07:
Relevant: https://github.com/ToyKeeper/anduril/pull/15

MacOS doing MacOS things, I guess.

ToyKeeper, 2023-11-07:
Can you use zsh?

I prefer zsh, but tried to write the script using only bash-isms because I thought it would already be installed and was unlikely to break.  But I guess the feature I wanted to use isn't as reliable as I thought.  Switching to zsh should fix it, but at the expense of adding a  build dependency which is less likely to be pre-installed.

ToyKeeper, 2023-11-07:
... or is this just a matter of finding the correct bash executable?  Like, would using "env bash" fix it?  (i.e. system bash is too barebones, but brew bash works fine)

gretel, 2023-11-07:
this is not a macos issue, but a portability one.

SiteRelEnby, 2023-11-07:
@ToyKeeper using /usr/bin/env bash should fix it. I think that bash is a better option than zsh as it will actually be available everywhere as well as being much better understood for scripting - MacOS just by default attempts to force zsh.

ToyKeeper, 2023-11-07:
I almost always write scripts in zsh unless I have a specific reason to use something else, because it has a lot of really useful features.  So I'm not as familiar with bashisms.  However, I don't mind using bash, since it has a nice set of features too.  Using plain /bin/sh (i.e. dash) makes a lot of simple operations unnecessarily complicated though, so I don't want to strip things down that far.

Anyway, I'll switch to env and try to fix the lint warnings... and hopefully that'll fix things on MacOS?

gretel, 2023-11-07:
> Anyway, I'll switch to env and try to fix the lint warnings... and hopefully that'll fix things on MacOS?

sounds good to me. the lowest common denominator is `sh`, tho.

maybe have a look at [`elv.sh`](https://elv.sh/) 😆

SiteRelEnby, 2023-11-07:
Yeah, POSIX sh is just an exercise in writing boilerplate and triple-checking everything. dash is somewhere between the two but doesn't have the same cross-platform compatibility.

Can confirm via work laptop that switching to /usr/bin/env bash works fine (something I tend to always do automatically, I guess).

See https://github.com/ToyKeeper/anduril/pull/15

ToyKeeper, 2023-11-07:
MacOS build fixed in bbdb68a04efb20b0c772b847706560d1b550e61b, I think.

Lint checks fixed in 96bb6aeb178182c95558c5157b6ddfaecfceb785, aside from a couple places where `shellcheck` was wrong.

closed 2023-11-07 by ToyKeeper

gretel, 2023-11-07:
> Lint checks fixed in [96bb6ae](https://github.com/ToyKeeper/anduril/commit/96bb6aeb178182c95558c5157b6ddfaecfceb785), aside from a couple places where `shellcheck` was wrong.

cool, there also is [`bashate`](https://github.com/openstack/bashate) which has a fun name and does similar but different 🔪 💻