ToyKeeper/anduril #20: Collection of ideas and improvements for aux handling

SammysHP, 2023-11-07:
For my own purposes I had refactored some aux related code. This includes:

- (previously an `indicator_led_update()` function, which is now already included upstream)
- refactoring of aux LED config to support more blinking patterns instead of only four
- more blinking patterns: simple low blink, simple high blink
- disable aux LEDs, but keep on button LED
- some improvements when aux LEDs are turned on (@ToyKeeper we shortly talked about it and you said that some of those were intentional)

https://github.com/SammysHP/flashlight-firmware/compare/multi-channel...mc-more-aux-patterns

Maybe you find some helpful modifications in this collection. If you have some roadmap in your mind, I could refactor some of my changes and make them mergeable.

lukestein, 2023-11-20:
Just a user endorsement of the value of a low blink. This allows having two low-only aux modes (steady low and low blink) to distinguish off from lockout on lights without rgb aux while protecting dark-adapted vision and battery life. Would love to see this in the main builds 🙏

SiteRelEnby, 2023-11-20:
FWIW I specifically like the idea of a high blink, or also constant-on low with the occasional high blink, which I think would be the best for locating a light in the dark, with the blink for if the aux are facing away so harder to see on low (but without them spending time completely off, also for some reason I find the current high/low blink animation stressful to watch, although can't really explain why). Plus, low with high blink would be great on disco mode.

SammysHP, 2023-11-21:
Indeed. I've added it to my fork as well, but with so many modes 7C gets very annoying. I think we need a menu to configure aux LEDs. Something like 7H enters the menu, items are mode, pattern and maybe more aux related settings.