ToyKeeper/anduril #25: FW3X low mode 2023-12-03

Light-Veteran, 2023-12-04:
Hi 👋🏻! Is fantastic you are here in GitHub! 
At level 1 there is a lot of flickering. I check for a decent level without any flicker and level **30** is good. Under this level light is with flicker. 
I used firmware without swaps RGB. Maybe I have to use the other firmware?
A big hug

ToyKeeper, 2023-12-08:
The reason for the ripple is because I made moon much, much lower than it was originally. The original firmware went down as far as it could without visible ripple, and made no attempt to go any lower. So moon was pretty bright. It also had really blocky steps at the bottom of the ramp, and the lowest levels weren’t very efficient because it kept the MCU running at full speed.

I changed it to go as low as possible, even though it’s unstable at those lower levels. Moon is much lower, and the low part of the ramp is much smoother, and the low modes have at least 4X as much runtime now… but it also means the driver’s noise floor and preflash are visible.

This is documented (at least somewhat) in the [readme](/hw/lumintop/fw3x-lume1/README.md).

The user can raise the moon level if desired, to stick to only the brighter / more stable levels… or live with the ripple in order to get lower lows. The original version didn’t give the user this choice, and instead opted to just remove unstable levels entirely.

ToyKeeper, 2023-12-08:
The delay before Batt Check is a new thing I changed on purpose.  It waits for a second before taking a reading, to give the measurement time to stabilize.  This fixed a problem Anduril has had since the beginning, where the first readout would give a different value than all the other readouts afterward.

At some point, I should probably try to reduce the delay, but that would require rewriting how the ADC code switches between voltage and temperature, and that would increase the code size and potentially bump some attiny85 builds over the limit... so for now I just used the bare minimum delay necessary to get a stable reading.

closed 2023-12-08 by ToyKeeper

Light-Veteran, 2023-12-08:
_"This fixed a problem Anduril has had since the beginning, where the first readout would give a different value than all the other readouts afterward."_

I understand and thank you for solve.
In this case the delay is necessary. Probably is my eye glitch after I see the delay for the first time! Don't remove the delay if is useful for correct voltage readout.