ToyKeeper/anduril #27: Reverse polarity warning / undefined behavior

singingtelegram, 2023-12-07:
Hi ToyKeeper! Big fan of your work.

I have a Noctigon M44 that I just flashed the 12-03 firmware, and it was working fine. However, during one charging session, I accidentally reversed the polarity when installing the cells (--/+). The voltage as measured from the top of the battery tube is ~4.18V (as opposed to ~12V in the correct series config)

I encountered some unexpected behaviors from the flashlight. 3C shows 135, which I later realized meant 1.35V. 1.35*3=4.05V, and after the voltage offset the read voltage should be around 1.40V (1.40V*3=4.20V, about right). However, since I did not realize that I installed the cells incorrectly at the moment, I considered the output uninterpretable. I thought the firmware or user config somehow got corrupted, so I attempted to reset the flashlight via 13H, but it only resulted in a single blink. Version check functioned as expected, and showed 0144/2023/12/03.

The light appeared to function normally and brightness levels were not impacted during my brief testing. However, I'm not sure if there should be some kind of LVP given that the MCU read an abnormally low voltage. It would also be helpful if a warning is implemented to alert the user that the cells were incorrectly installed.

The light functioned normally after the cells were installed correctly.

ToyKeeper, 2023-12-08:
The battery check mode shows the voltage per cell.  If a light has 3 cells in series and the total voltage is 4.18, the average per cell would be 1.39V.  It looks like the reading of 1.35V was a bit low, but not by much.

Since the M44 uses a boost driver, it's not too surprising that it would still work at a fraction of its intended input voltage.  The boost converter would simply need to work extra hard, and would get hot at a significantly lower output level than usual.  But I doubt it would have time for thermal regulation to activate, because LVP would activate first.

It *does* have LVP.  Like, when you attempted to factory reset and it failed, that was probably the LVP kicking in and forcing the light to shut off.  It would have also reduced brightness every few seconds during normal use, and then shut itself off after reaching the lowest level.  The way LVP works is:

- In ramp mode, reduce power by about half.  If already at the lowest level, shut off.
- In other modes, either switch to ramp mode at a low-ish level or shut off immediately.

These actions continue every 5 seconds or so until voltage recovers to an acceptable level, or until the light is off.

LVP activates at 2.9V per cell, which is low enough to use a battery's full charge, but high enough to catch the case when a user has 2 good cells and 1 dead cell (i.e. 2.8V avg)... or when they put a cell in backward (1.4V avg).

I hope nothing was damaged.  Running with two cells in direct opposition is pretty dangerous.

ToyKeeper, 2023-12-09:
The MCU doesn't have enough information to tell the difference between "one or more cells inserted wrong", "one or more cells dead", and "battery is really low"... so it treats them all with the standard LVP handler.

Is there anything which needs to be changed, or is it sufficient to step down and shut itself off like it does now?