ToyKeeper/anduril #29: Wurkkos TS11 multichannel not working properly

diRk262, 2023-12-10:
Hi,
I updated my flashlights to the latest version 2023-12-03.
Unfortunately I can't get the rgb multichannel working on my TS11.
My TS10rgb is working fine so I guess it is a bug.

In andavlnced UI when the flashlight is ON:
9H until the red leds light up, release button, confirm the fast blinking wirh one click. 

On my TS10 rgb I can now switch between the normal led and the red one with 3C when the light is on.
On my TS11 with 3C the main led turns off but the blue one turns on (maybe voltage color).

I don't think is how it is supposed to be. 

Thanks

ToyKeeper, 2023-12-11:
When using an aux channel as the main output for ramp mode, the colors are wrong because it's trying to display battery voltage on the RGB button... and the RGB button is physically connected to the same wires as the front RGB aux, so they are locked to the same colors.

At the moment, there are two options:

- Use the TS25 firmware.  This allows RGB aux to be used in ramp mode, but it means the button won't show battery voltage during operation.
- Use the TS11 firmware.  This shows battery voltage while on, but RGB aux doesn't work in ramp mode.

The only difference between these two builds is whether RGB aux shows battery status while the light is on.

At some point, it should probably try to detect whether the light is in ramp mode on a RGB aux channel, and temporarily disable the RGB voltage function...  but it doesn't do that yet.

SiteRelEnby, 2023-12-11:
I wrote a feature that did this on my old fork. I'm dealing with some personal life stuff rn so don't have time yet, possibly tonight or tomorrow evening, but if you want to just take what's already there it's there, otherwise I'll try and have a go at porting it to the refactor.

diRk262, 2023-12-11:
> * Use the TS25 firmware.  This allows RGB aux to be used in ramp mode, but it means the button won't show battery voltage during operation.

Oh I didn't knew that I can switch between these 2 firmwares. But I like the rgb voltage for the button, so I will stay with the TS11 fw and disable the rgb multichannel (for now).

> At some point, it should probably try to detect whether the light is in ramp mode on a RGB aux channel, and temporarily disable the RGB voltage function... but it doesn't do that yet.

I thought this would be the way it works already. Thats why I opened the issue.

Maybe it will be possible to deactivate the rgb voltage button feature as soon as a rgb led is enabled for multichannel in a future release.

diRk262, 2023-12-11:
> I wrote a feature that did this on my old fork. I'm dealing with some personal life stuff rn so don't have time yet, possibly tonight or tomorrow evening, but if you want to just take what's already there it's there, otherwise I'll try and have a go at porting it to the refactor.

Thank you. I will take a look at it.
Just take your time and don't be in a rush. My flashlights are working and this is just a little "bug" which is not very hard to live with :)

SiteRelEnby, 2023-12-17:
@ToyKeeper Done.

gretel, 2023-12-21:
works fine on my TS11.