ToyKeeper/anduril #30: newest FW for Noctigon KR4 with boost driver not working

q-l-r, 2023-12-12:
After reflashing with 2023-10-31 FW (noctigon-kr4-boost) and inserting a battery the flashlight spontaneously ramps up, ramps down, then flashes once, and begin flickering on a low mode. The switch does not respond. After reflashing with an older FW it works fine.

frzolp, 2024-04-30:
I am also having this issue, but with a KR1 driving an FC-40 (Anduril model number `0216`). It works fine on the 2022-10-21 build of `noctigon-kr4-12v` but subsequent builds of `hank-noctigon-kr4-boost` show the above symptoms of ramping up, ramping down, flashing, then flickering with an unresponsive button. Unlike OP, flashing `hank-noctigon-dm11-boost` results in the light not turning on and the switch not responding. Flashing the KR1 back to the 2022-10-21 build returns it to normal working order.

SiteRelEnby, 2024-04-30:
My KR1 with FC40 uses 0273 (dm11-boost) so it seems like Hank has used each driver at different times, as they use a different switch pin. That's really weird though as *one of* those should have worked, the only difference in the hardware config is which pin the MCU uses for the switch, and in both drivers then the opposite pin is left unconnected so it shouldn't be getting read as a spurious input from the wrong firmware. Did the firmware verify correctly after flashing?

frzolp, 2024-04-30:
Yep, confirmed flashing was successful and verified each time. I can triple-check again later, but I had the same issue on 2023-12-03 and 2024-04-20 builds and had the same behavior from `hank-noctigon-kr4-boost` (ramping up/down, unresponsive button) and `hank-noctigon-dm11-boost` (no light/no button). 

Driver PCB is purple, there's a blue dot in the center where the positive terminal spring is, if that helps.

frzolp, 2024-04-30:
Just repeated flashing the 2024-04-20 builds of `hank-noctigon-kr4-boost` and `hank-noctigon-dm11-boost`, same as before. 

Tried holding down the power button while tightening the cap on both FWs to see if I could get it to reset. The KR4 build did the normal reset strobe pattern, but then I couldn't get the light to respond to any subsequent button presses. The DM11 build didn't respond to the factory reset hold or any other button actions.

Flashing `anduril.2024-04-20.hank-noctigon-kr4-boost.hex` using Hank's usbasp programmer and adapter.
```
avrdude.exe -p t1634 -c usbasp -u -U flash:w:anduril.2024-04-20.hank-noctigon-kr4-boost.hex
avrdude error: "safemode" feature no longer supported
avrdude error: cannot set sck period; please check for usbasp firmware update
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9412 (probably t1634)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude error: cannot set sck period; please check for usbasp firmware update

avrdude: processing -U flash:w:anduril.2024-04-20.hank-noctigon-kr4-boost.hex:i
avrdude: reading input file anduril.2024-04-20.hank-noctigon-kr4-boost.hex for flash
         with 10612 bytes in 1 section within [0, 0x2973]
         using 332 pages and 12 pad bytes
avrdude: writing 10612 bytes flash ...
Writing | ################################################## | 100% 8.57 s
avrdude: 10612 bytes of flash written
avrdude: verifying flash memory against anduril.2024-04-20.hank-noctigon-kr4-boost.hex
Reading | ################################################## | 100% 6.47 s
avrdude: 10612 bytes of flash verified

avrdude done.  Thank you.
```

Flashing `anduril.2024-04-20.hank-noctigon-dm11-boost.hex`
```
avrdude.exe -p t1634 -c usbasp -u -U flash:w:anduril.2024-04-20.hank-noctigon-dm11-boost.hex
avrdude error: "safemode" feature no longer supported
avrdude error: cannot set sck period; please check for usbasp firmware update
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9412 (probably t1634)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude error: cannot set sck period; please check for usbasp firmware update

avrdude: processing -U flash:w:anduril.2024-04-20.hank-noctigon-dm11-boost.hex:i
avrdude: reading input file anduril.2024-04-20.hank-noctigon-dm11-boost.hex for flash
         with 10640 bytes in 1 section within [0, 0x298f]
         using 333 pages and 16 pad bytes
avrdude: writing 10640 bytes flash ...
Writing | ################################################## | 100% 8.61 s
avrdude: 10640 bytes of flash written
avrdude: verifying flash memory against anduril.2024-04-20.hank-noctigon-dm11-boost.hex
Reading | ################################################## | 100% 6.48 s
avrdude: 10640 bytes of flash verified

avrdude done.  Thank you.
```

And flashing the light back to `anduril.2022-10-21.noctigon-kr4-12v.hex`
```
avrdude.exe -p t1634 -c usbasp -u -U flash:w:anduril.2022-10-21.noctigon-kr4-12v.hex
avrdude error: "safemode" feature no longer supported
avrdude error: cannot set sck period; please check for usbasp firmware update
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9412 (probably t1634)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude error: cannot set sck period; please check for usbasp firmware update

avrdude: processing -U flash:w:anduril.2022-10-21.noctigon-kr4-12v.hex:i
avrdude: reading input file anduril.2022-10-21.noctigon-kr4-12v.hex for flash
         with 9502 bytes in 1 section within [0, 0x251d]
         using 297 pages and 2 pad bytes
avrdude: writing 9502 bytes flash ...
Writing | ################################################## | 100% 7.67 s
avrdude: 9502 bytes of flash written
avrdude: verifying flash memory against anduril.2022-10-21.noctigon-kr4-12v.hex
Reading | ################################################## | 100% 5.79 s
avrdude: 9502 bytes of flash verified

avrdude done.  Thank you.
```