ToyKeeper/anduril #32: Add option to skip DFP check

SammysHP, 2023-12-13:
The tinyAVR 0/1 series are supported by avr-libc (development version, no release yet), thus the "proprietary" Atmel DFP isn't necessary.

In the past one was able to fake its existence by setting an environment variable. The new build script sets that variable by itself and makes it impossible to override the location.

ToyKeeper, 2023-12-15:
Does commit 0725045fc2d6f974f50d59755f9bbecd0c33ee23 fix this issue?

If I understand correctly, it should be sufficient to skip the "make dfp" step during setup, since the system libraries should provide the necessary headers.

SammysHP, 2023-12-15:
Yes, it works. Actually that's more than I asked. My idea was something like an environment variable like `DISABLE_DFP_CHECK`, that I can add to my Docker image.

SiteRelEnby, 2023-12-15:
Could always merge this one ;)

https://github.com/ToyKeeper/anduril/pull/12

gretel, 2023-12-20:
in the meantime https://github.com/ToyKeeper/anduril/commit/0725045fc2d6f974f50d59755f9bbecd0c33ee23