ToyKeeper/anduril #34: thefreeman-avr32dd20-devkit has dubious model id

gretel, 2023-12-20:
https://github.com/ToyKeeper/anduril/blob/312f7442c31d72dd36f108de9542f81960526658/MODELS#L79

ToyKeeper, 2023-12-20:
The model ID is intentional.  I added support for hex digits in Version Check, to make it possible to use model numbers like this.  The change also makes it possible to include git commit hash references in the version number.

closed 2023-12-20 by ToyKeeper

gretel, 2023-12-25:
maybe it's my lack of understanding but i don't see why it has to be done this way, or with any change to the actual codebase at all. i think the github actions way is in widespread use among similar projects because it can be very simple and therefore robust to have even complex and complicated release patterns reflected.