ToyKeeper/anduril #35: No ceiling to be found at all in smooth ramp style

fredrik1102, 2023-12-20:
@ToyKeeper 
Hello there. 
I have a "Sofirn Q8 Plus" which when I got it delivered flashed out version "2022-02-08--0614" 
In other words the "Sofirn Q8 Plus" uses "sofirn-sp36-t1616" firmware. 
(It was I who asked you here) https://budgetlightforum.com/t/sofirn-q8-plus/216834/110

I have flashed my "Sofirn Q8 Plus" with your precompiled "anduril.2023-12-03.sofirn-sp36-t1616.hex"
I have also compiled my own slim down version from your latest source which doesn't have simple UI included and some other things I don't need, same results.

The problem is in advanced UI when smooth ramp style is chosen, from Off, 2C goes to turbo (should have gone to ceiling) 
On, 1H ramps right up to turbo without stopping at ceiling, and 2C here does nothing, if I 3C here to stepped ramp style, now 2C works and toggles between turbo and ceiling.
No ceiling to be found at all in smooth ramp style. 
But everything in stepped ramp style works as it should, it's only smooth ramp style that misbehaves.

SammysHP, 2023-12-20:
It is not misbehaving, it is just defined this way:

https://github.com/ToyKeeper/anduril/blob/312f7442c31d72dd36f108de9542f81960526658/hw/sofirn/blf-q8-t1616/anduril.h#L35-L40

Smooth from 1 to 150, stepped from 20 to 130.

ToyKeeper, 2023-12-20:
I think Sofirn asked to have the ceiling set to full power in smooth ramp mode.  The user can set it lower if desired, but I think they just wanted to make it easy for people to reach full power.

closed 2023-12-20 by ToyKeeper

gretel, 2023-12-21:
> I think Sofirn asked to have the ceiling set to full power in smooth ramp mode. The user can set it lower if desired, but I think they just wanted to make it easy for people to reach full power.

why do vendors have that sort of power to decide what is "best"?