ToyKeeper/anduril #40: Tactical Mode 3C cycles through the channels but gets stuck in Channel 1

zaziboo, 2024-01-15:
Tactical Mode 3C cycles through the channels and when it gets to Channel 1, gets stuck there. Successive 3C no longer works.

Setup:
Flashlight: Dual Channel KR4 (HW: 0215)
Channels enabled (On -> 9H): Ch0, Ch1, Ch2
Strobe mode in Tactical uses all 4 LEDs.

Work-around (if you're a channel switcher and a tactical mode user that doesn't want to end up with Ch1 only during Tac mode):
Disable all the channels except Ch3 (manual-blend, 4th blink during On->9H)
Set Misc Config to 3 steps (Off->9H)
Channel switching now becomes a 3H (Ch1, Both, Ch2 quick switch)
Set the right channel first in ramp mode before going tactical mode.

dirtydancing, 2024-01-15:
@zaziboo  Which Anduril 2 version are you using?

I just checked with Emisar D2 (dual channel), running ToyKeeper's latest release, version check `0135-2023-12-03`, and I can *not* reproduce this issue. In Tactical Mode, with 3C, I can switch through all enabled channel modes (be it the default 5 or only 3) round and round, and at no time Tactical Mode gets stuck in one specific channel mode.

And just to exclude this possibility: Do you happen to have Manual Memory activated?

I am asking because with multi-channel lights, Manual Memory will save the current brightness as well as the current channel mode. This also applies to Tactical Mode. Once you wait a bit in "Off" state after attempting to switch the channel mode with 3C in Tactical Mode, the manually memorized channel mode will come up, and this may lead to the impression that you are "stuck".

zaziboo, 2024-01-16:
I'm also using the latest Dec release, 0215-2023-12-03. Unfortunately, I don't have a D2 to compare against (but I have a dual channel DT8K incoming, which I'll do another test with.) 

Your 3C cycle works regardless of which channel you start Tactical mode? 

Yes, I have 5-min manual memory mode enabled (for all 4 LEDs, ie enabled for Ch2.) Let me redo the test without it and report back.

zaziboo, 2024-01-16:
UPDATE: 
The issue happens once you change the Tactical Slot 3 to option 152. After that, once you 3C, it switches to the next channel and then gets stuck. Then 3C no longer shifts channels.

Other observations:
A. Setting Tactical Slot 3 to 0. 6C back to normal mode:
1. Off -> 3H and go to Police Strobe. Off -> 6C (where 3H becomes Police Strobe) : No issues with 3C
2. Switching to other Strobe/Mood Modes will result in Tactical Mode 3C to get stuck

dirtydancing, 2024-01-16:
Issue reproduced!

Issue reproduced with Emisar D2 after setting Tactical Slot 3 to Tactical Strobe (152). I will so some more testing and report back; this might be a little while.

dirtydancing, 2024-01-17:
Reproduced with D2 dual channel what OP reported for KR4 dual channel. For my observations with the D2, I am using the following terminology, based on [hwdef.h for emisar-2ch](https://github.com/ToyKeeper/anduril/blob/76ce3c5e70bf415fbf28597db994ca04416e7ed3/hw/hank/emisar-2ch/hwdef.h#L39C1-L53) (the first five channel modes are enabled by default, not the Aux LED Channels, `Ch5+`, sixth+ channel modes):

`Ch0` - first channel mode (Channel 1)
`Ch1` - second channel mode (Channel 2)
`Ch2` - third channel mode (50/50)
`Ch3` - fourth channel mode (Tint Ramp)
`Ch4` - fifth channel mode (Autotint)

After Factory Reset (physical) and 10H from Off to switch from Simple UI to Advanced UI, 6C from Off leads to Tactical Mode. Tactical Slot 3 by default is Police Strobe. 3C switches through the 5 channel modes, starting from `Ch3`. Channel mode changes in Tactical Mode carry over to regular Ramp/On mode, once Tactical Mode is left with 6C again.

No matter which channel mode is switched to with 3C in Ramp/On, switching through all channel modes in Tactical Mode with 3C is always possible.

Starting from `CH0` in Ramp/On, entering Tactical Mode with 6C from Off, and then with 7H in Tactical Mode setting Tactical Slot 3 to Tactical Strobe, 152C, leads to the present issue: Switching channel modes in Tactical Mode gets stuck in `CH1`. 1H and 2H in Tactical Mode are `Ch1`, but 3H, Tactical Strobe, is `Ch0`. After leaving Tactical Mode with 6C, for Ramp/On mode `Ch1` is set, but 3C for switching channel mode is working just fine in Ramp/On mode.

Starting e.g. with `Ch2` in Ramp/On, entering Tactical Mode leads to 1H and 2H with `Ch2`, but 3H, Tactical Strobe, is `Ch0` again. And switching channels with 3C does not lead to `Ch3`, but to `Ch1` again, again with `Ch1` for 1H and 2H, and `CH0` for 3H, Tactical Strobe. And again, when returning to Ramp/On mode, `Ch1` will be set, but 3C for switching channel mode is working just fine in Ramp/On mode.

As there would be too many possible testing combinations with all of these variables, I then restricted my testing to setting Tactical Slot 3 to 0C, for the last-used strobe / mood mode, after another Factory Rest (physical). And I could reproduce what OP has already reported: All strobe / mood modes except for Police Strobe lead to the issue of getting stuck in `Ch1` in Tactical Mode when attempting to switch channel modes with 3C, for details cf. above.

One more observation: When setting Tactical Slot 3 to 0C, last-used strobe / mood mode, then using Tactical Strobe in the strobe / mood modes and with 3C switching the channel mode for Tactical Strobe (possible per strobe mode) from `Ch0` to `Ch3`, then going back to Tactical Mode, this happens: In Tactical Mode, 3H will be Tactical Strobe with `CH3`; but now trying to switch channel modes in Tactical Mode with 3C appears not to work at all, the light just keeps staying at both channels, it does not switch to `Ch1`, and obviously, in this setup, it does not get stuck in `Ch1` either, as apparently it already gets stuck before. In any event, 3C for switching channel modes in Tactical Mode is not working here, either.

Looking at all this, maybe there is a connection between 3C in Tactical Mode working when Police Strobe is enabled, but not working with all other strobe / mood modes, and 3C in the strobe / mood modes: Police Strobe is the only strobe / mood mode in which 3C for switching channel modes per strobe / mood mode is not available. So maybe the ability to switch channel modes per strobe / mood mode with 3C in all the other strobe / mood modes is influencing the ability to switch channel modes in Tactical Mode with 3C?

ToyKeeper, 2024-02-03:
Ah, interesting.  In tactical mode, 3C should not be changing the channel at all.  The 3C and 3H actions are supposed to only cause the 3rd mode to run, not to run the 3rd mode *and* change the channel.  So it probably needs an extra clause to swallow the 3C input instead of letting it get passed to another function.

dirtydancing, 2024-02-04:
> In tactical mode, 3C should not be changing the channel at all.

Thank you for clarifying this. I was assuming that in the UI Reference Table at the end of the Anduril User Manual, [`3C` for next channel mode](https://github.com/ToyKeeper/anduril/blob/cbfc1a13a9c9a8c08c054658ed26d2350383d209/docs/anduril-manual.md?plain=1#L1025) in Multi-channel Lights for `Any Mode` was also covering Tactical Mode. Upon closer inspection, I have come to realise that the relevant point is that the `3C` button mapping is not listed in [Tactical Mode](https://github.com/ToyKeeper/anduril/blob/cbfc1a13a9c9a8c08c054658ed26d2350383d209/docs/anduril-manual.md?plain=1#L1107), in contrast e.g. to the [Strobe Group Modes](https://github.com/ToyKeeper/anduril/blob/cbfc1a13a9c9a8c08c054658ed26d2350383d209/docs/anduril-manual.md?plain=1#L1052) (and that moreover, `3C` for next channel mode in Multi-channel Lights for `Any Mode` is for `Any UI`, whereas Tactical Mode is for `Full UI` only).

SiteRelEnby, 2024-02-14:
https://github.com/ToyKeeper/anduril/pull/49

ToyKeeper, 2024-03-28:
Fixed and merged.  Thanks!

closed 2024-03-28 by ToyKeeper