ToyKeeper/anduril #41: Sofirn sc21 pro show wrong firmware?

DeddyIta, 2024-01-18:
Thank you for your hard work!
I bought a sofirn sc21 pro, and it had on an old firmware from 2022. 
I updated my sc21 pro to the latest firmware 0632-2023-12-03. 
Everything works fine, but when I do 15 clicks to check the version, the flashlight show 0632-2023-10-31.
[image]

ToyKeeper, 2024-02-03:
Are you sure you got the right file?  I flashed that file to a light just now, and version check show the 12-03 date code.

A quick way to tell is to use the battery check mode.  If it starts blinking out a value immediately, and gives 2 digits of precision, that's an older version.  If it pauses for a moment before the readout, then gives 3 digits of precision, that's 2023-12-03 or later.

DeddyIta, 2024-02-03:
Thanks for the reply ToyKeeper. I'm sure I've tried several times to download the latest anduril version from the repo and use the hex file specific to the sofirn sc21 pro. I also tried the battery reading and it gives me 2 digits.I don't have any other anduril folders in my smartphone other than the one of the latest version.

DeddyIta, 2024-02-04:
I tried to download the folder again, and proceeded to flash the flashlight firmware again. Now show the right version. Thank you.

closed 2024-02-04 by DeddyIta