ToyKeeper/anduril #42: FW3A does not have tactical mode

fiveseven808, 2024-01-20:
Running firmware `311-2023-12-03`

Steps to reproduce: 
- Flash firmware
- 10C from off to Advanced UI
- 6C from off to Tactical

Result: Light does nothing, no tactical mode, still in ramp mode

Expected result: Tactical mode 

A bit disappointed as tactical mode is my new favorite mode and has me finally truly moving off of surefires with my other lights. The FWXX would be amazing to have tactical model on. Thank you for all of your hard work!

SammysHP, 2024-01-20:
The FW3A has an older driver with not enough memory for all features.

fiveseven808, 2024-01-20:
Any idea how I can go about make a special build with features I don't care about but add in tactical mode? I see in the fw3a anduril.h on line 51 there's `#undef USE_TACTICAL_MODE`

Looks like I can comment that out, but I'm not sure how to figure out what other features I can disable, as well as figure out if the compiled build will fit on the FW3A?

fiveseven808, 2024-01-20:
I think I figured out what I can #undef in anduril.c lines 157ish to 218ish? 

Any idea how to figure out what particular size I'm shooting for?

SiteRelEnby, 2024-01-20:
The easiest way is just to build it, and the build will fail if it's over the limit.

If you fork it in github you can use the CI workflows to build as well, or can use https://github.com/SiteRelEnby/anduril-buildenv-docker.

In general stuff I usually find good ones to get rid of are SOS mode, simple UI, any other strobes you don't necessarily need (party strobe, lightning storm, etc), but there are lots of other small features you can play with, in general removing SOS mode should get you most of the way there on most lights.

SammysHP, 2024-01-20:
The FW3A needs a lot of memory for the ramping tables because it has three main channels. Thus you have to remove some more features. Sunset mode might be a good candidate, also all the fancy strobe modes.

fiveseven808, 2024-01-20:
@SiteRelEnby Thanks! Exactly what I need to know

@SammysHP  That makes sense. I was trying to compare with the D4 which looks like it still has Tactical Mode enabled. Thanks!

Question, I also have a FET + 1 FW3A that came with Anduril 2 and appears to be have come with the BLF GT Mini firmware. Since it only has 2 main channels like the D4, is there any obvious reason why Tactical mode was omitted in the anduril header for the BLF GT Mini?

fiveseven808, 2024-01-22:
Ok so I was able to flash both my FET+1 FW3A and FET+7+1 FW3A with tactical mode!

FET+1 definitely ran the BLF GT Mini firmware. I actually enabled everything in the blf-gt-mini anduril.h and removed the defines for the lighted button. This thing is running everything but SOS mode! I suppose that's one thing this "gimped" FW3A is good for? Max for 1x7135 is level 65 it seems? level 65 is fine for most tasks around the house. I don't really miss the 7x7135s everyone else whines about. I'm normally not in that level, and if I am its not very often. My tactical mode levels are set to 150 and 65 because of this new information.

On the FET+7+1 SOS was removed by default, but it still didn't want to compile with tactical mode. 
I removed the sunset timer and that was enough to make space for tactical mode. I certainly never use it. Not entirely sure what the purpose of it is for. Night light for kids?

Anyway, I'm happy that my FW3A/FW1As are getting massively upgraded IMO! 

Even though I've solved my own personal problems moving forward I'm going to leave this issue open for the following reasons:
- Current version of anduril is available to a wide range of lights, but not all features are available on all lights and this does not appear to be documented
- Even if it was known that not all features are available on all firmwares, it's impossible to know what features have been removed unless you look at the code
- I propose that at minimum a README file gets put in the directory of all lights that have less than full features explaining what features have been cut
- It would also be good to build out the compiling section to help streamline making custom anduril firmwares. It was a fairly simple process to use WSL for me, but I also have a little bit of experience with WSL. 

I can't program very well but I'd be interested in building out a document trying to outline how a windows user can modify and compile firmware? That'd be kind of cool! There doesn't seem to be a generic "how to modify anduril" guide yet!

SiteRelEnby, 2024-02-03:
Yeah, it's something I've had on my mind for a while, question of finding the time and the spoons.

ToyKeeper, 2024-02-03:
> is there any obvious reason why Tactical mode was omitted in the anduril header for the BLF GT Mini?

To make enough room for the firmware to fit.  The GT-Mini takes extra space to add support for the lighted button... which doesn't even exist on FW3A.  So if you need more room, remove the lighted button features.  Lumintop didn't follow the proper dev procedures for a lot of their lights, so they ship with firmware which was never intended for those lights.

You already figured this out though.

> I propose that at minimum a README file gets put in the directory of all lights

I agree.  And ideally, I'd like to generate some of the config info automatically, to ensure it stays in sync with the actual code.  I'm thinking of adding something to the compile process so that, after the preprocessor generates the full merged code file, something could parse that to determine the state of various options, and generate a summary in a consistent format.  It already generates that post-processed source file... it just needs an extra script to pull relevant options out of it and dump them into a new file which is easier to read.

Although not the original intent of this bug, I'm going to re-title it to indicate that the compile script needs to generate a config summary.  This could then be included in the readme or in a feature table or ... whatever.  Details for that are yet to be determined, but the data should be available in a format which is simple to parse and read.