ToyKeeper/anduril #44: Question: Can someone explain Aux brightness levels?

fiveseven808, 2024-01-23:
So, I'm under the understanding that AUX lights aren't PWMed to achieve the lower brightness because that would keep the MCU awake. 

I can't find any schematic for any light showing the AUX leds other than just some lines off of the MCU dedicated to them. 

I assume they're driven directly off of the MCUs output lines through a resistor for "high"? But it's unclear to me how we're able to get the second "low" brightness level. 

Can someone explain this to me? 

Reason for asking:
I want to bridge the resistor on the RED AUX channel on a D4v2, and I want to know if I'm gong to fry the MCU or affect the "low" aux value, and if so by how much?

SammysHP, 2024-01-23:
> I assume they're driven directly off of the MCUs output lines through a resistor for "high"?

Correct.

> But it's unclear to me how we're able to get the second "low" brightness level.

The MCU contains internal pull-up resistors. For the low mode it is enabled in series with the external resistor.

> I want to bridge the resistor on the RED AUX channel on a D4v2, and I want to know if I'm gong to fry the MCU or affect the "low" aux value, and if so by how much?

The low mode won't be affected much, the high mode could damage the aux LED and/or the MCU.

fiveseven808, 2024-01-25:
Thank you, that clarifies a lot. Looks like I'd need to replace the resistor for what I want to do.

closed 2024-01-25 by fiveseven808

Metadorius, 2024-08-26:
> The MCU contains internal pull-up resistors. For the low mode it is enabled in series with the external resistor.

@SammysHP per my understanding, this means that I can't make the low level AUX / button lights on my D3AA dimmer without physically changing the external resistors to be of higher resistance, am I correct?

loneoceans, 2024-08-26:
> @SammysHP per my understanding, this means that I can't make the low level AUX / button lights on my D3AA dimmer without physically changing the external resistors to be of higher resistance, am I [...]

That is correct. Anduril on the D3AA uses the internal pull-up resistors of the AVR32DD20 pins for the low aux-mode, which are in series with the external aux resistors. Unfortunately, this internal pull-up resistor value is not well controlled (it's in fact, not meant to be used as a current source which Anduril does), and the datasheet value suggests that it can range from ~15k to 33kR. Changing the external resistors (to be larger) will have a larger effect of changing the high-level Aux more than the low-level, but that would be the only way. I have a different implementation for aux LEDs for a potential future driver whereby aux brightness can be dynamically controlled without too much power cost, but it's unclear if it will make it into Anduril since it requires additional hardware.