ToyKeeper/anduril #45: 150/150 brightness PWM question

fiveseven808, 2024-01-25:
So, I'm looking at the FW3A anduril.h and I noticed that the PWM levels at the 150th level, drop off to 0 for the 7135 channels, and just go to 255 on the FET Channel. 

Is there any reason why doing 255 to all 3 PWM channels isn't better? 

I agree that the FET _should_ be able to deliver max current if left at 100% PWM, however Is there a downside to running the 7135 channels at the same time??? 

The reason I'm asking this is that I have both the FET+1 and FET+7+1 types of FW3A drivers, and my FET+1 is noticeably dimmer on turbo than the FET+7+1.  Both are using stock emitters. Could be binning? 

I enabled 255 on the 1x7135 channel on the FET+1 for turbo but it either made things dimmer, or did nothing at all.

SammysHP, 2024-01-25:
The 7135 regulators can't operate if something else is "stealing all the power". They will be in an undefined state which could cause all kind of trouble.

Do you have the same emitters in both lights? Do you actually use 150/150? Turbo is heavily affected by battery voltage and small resistance here and there which adds up (wires, springs, FET).

ToyKeeper, 2024-02-03:
The DD FET is very low resistance, almost like connecting the LEDs directly to the battery with a wire.  The 7135 chips have higher resistance.  Connecting the FET by itself usually produces higher brightness than connecting all the 7135 chips at the same time, because the 7135 chips basically just add unnecessary resistance.

This effect is large enough it can frequently be seen by eye as soon as thermal regulation begins.  The drop from level 150 to 149 tends to be quite a bit larger than the drop from 149 to 148.  All those 7135 chips add quite a bit of resistance.

So it turns the 7135 chips off for turbo, in order to make turbo brighter.

About the FET+7+1 vs FET+1 drivers, there are a bunch of things which could explain the difference.  If the LEDs aren't the same type, that would cause large differences.  If the battery is a different type or charge level or age, that would make a pretty big difference too.  Turbo is mostly just the LEDs, the battery, and some parts between to carry the power... so the total brightness depends on those three factors.

It's also worth mentioning that Lumintop has changed many of the components between hardware batches, so it's very possible there's more resistance somewhere.  Consistency and quality control have been extremely poor for the entire Lumintop FW product line, since the very beginning.

Or it could just have some dirt or grease somewhere, or a loose retaining ring, adding resistance.

closed 2024-02-03 by ToyKeeper