ToyKeeper/anduril #48: Possibility of changing timers from "hold to add time", to "click to add time"?

maru801, 2024-02-14:
For both the "sunset timer" and "candle timer", would it be possible to change their behavior from a hold to add time, to instead be able to click x number of times to increase the time?

I was thinking it's weird that you have to hold to add 5 minutes per blink, when the rest of the UI relies on clicks and click-holds to add to things. I feel like it would be faster to set the timer to, lets say 1 hour, by click-holding 6 times (10 min per click-hold) like the rest of the UI. This could even be different to where a single click is 5 minutes too, so 12 clicks gets you to an hour and saves you the 8 extra seconds it would take otherwise.

SiteRelEnby, 2024-02-14:
That would make any other button combinations with a hold (e.g. 9H) impossible.

maru801, 2024-02-14:
Figured. There's no possibility of this due that compatibility issue then? Should I close this issue?
Btw, I didn't mean to post this as an issue, but we can't assign the "enhancement" label by ourselves.

SiteRelEnby, 2024-02-14:
If there were any other ways of implementing sunset mode it'd be interesting to hear - unfortunately if 5H can be taken as something that isn't desired to change then I'm not sure what else - it would be possible to use the config menu and number entry code to make a menu on 5H where you then enter minutes as the value, I guess, but not sure if that would be broadly preferred or not.

maru801, 2024-02-14:
Ok, so looking at the following code: https://github.com/ToyKeeper/anduril/blob/cbfc1a13a9c9a8c08c054658ed26d2350383d209/ui/anduril/sunset-timer.h#L7-L8

Would I be fine in changing the the sunset timer unit to 10? Would this mess up other things that rely on it? I assume this line only applies to sunset and candle modes only.

SiteRelEnby, 2024-02-14:
Yeah, that would work if you wanted to build anduril for your own lights with that change.

maru801, 2024-02-14:
Thanks for your response @SiteRelEnby .

I'll go ahead and mark this as closed.

closed 2024-02-14 by maru801