ToyKeeper/anduril #58: Request: adding thermal regulation to blinking modes

Carpe-Diem-Postero, 2024-04-05:
Hi. I know we shouldn't use our light is high output  for extended periods of time, but I also think that thermal regulation is a safety feature that should be available in every mode.

I burned myself (finger and thight, second degree) when grabbing a TS10 that I left in candle mode for bedside while reading. I didn't knew that blinking modes don't throttle thermally, so I just assumed it would step down naturally, as it does in ramping mode. The battery suffered severe heat, to the point the plastic cover shrank and now the lower third of it is exposed.

I know the shame's on me, as I should have checked the temperature manually. Granted, I'm the one to blame. But I also think that output thermal regulation is a safety feature that should be enabled in every mode, as it is dangerous, specially in bike or candle mode (I can imagine someone getting burned while  driving his bike).

 I saw that this issue is documentes for at least 3 years now. It is possible to enable this safety feature for blinking modes, or is there a hardware limitation?

SiteRelEnby, 2024-04-07:
There's no hardware limitation, more a "wanting to work on thermal regulation code" limitation, but I agree it should definitely be done.

dpse, 2024-04-19:
I've made an attempt in #66.

Carpe-Diem-Postero, 2024-04-19:
> I've made an attempt in #66.

Great! Thanks!

I just received my CH340, I don't know how to compile it, but if someone prepares the .hex, I can test it in lights: 2xTS10 RGB, TS10 single aux, SP10, TS11, noctigon KR4 and Emisar D4K

dpse, 2024-04-20:
That would be great! I've only tested with a D4K. [Here](https://github.com/ToyKeeper/anduril/files/15047463/hex.zip) are some hex-files. Make sure to back up your current firmware and check the model number to select the correct hex-file!

Carpe-Diem-Postero, 2024-04-20:
Perfect, I will begin test ASAP (depending of WAF, of course ^_^U).

Is there anything specific I should be looking for or test aside thermal step down in blinkies and in normal ramping mode?

Carpe-Diem-Postero, 2024-04-21:
> That would be great! I've only tested with a D4K. [Here](https://github.com/ToyKeeper/anduril/files/15047463/hex.zip) are some hex-files. Make sure to back up your current firmware and check the model number to select the correct hex-file!

One question: the TS10 hex file included in your zip is the rgb aux version or the single aux one?

dpse, 2024-04-22:
No not really. But let me know if you see anything strange in the blinkies. Candle mode is automatically scaled down so it should still flicker nicely even when stepped down but the other blinkies are just cut off so there is a probability of something looking weird with a light that does not have very good heat dissipation...

Carpe-Diem-Postero, 2024-04-22:
Ok. I will report back.

Yesterday I was making a quick test, and I noticed two things:

1) TS10 does not get as hot in candle as with the previous firmware, I think because maximum output is probably already capped down. So, it does not seems possible to achieve high temperature

2) bike mode can overheat, and light seems to step down brightness (I still need to confirm this using a luxometer app), but it does not step up again when temperature lowers. If you stop blink mode and restart it, brightness remains lowered. But if you turn off blinking and then turn on the light on normal ramping mode, I can see the light turning on in low output, and immediately stepping up "normal" output. That seems to "reset" the ceiling level, and bike mode seems to get bright again.

This is a preliminary "letmetrythisin5minutes" report. I will do my best to make a proper test with brightness measures and will report when I can.

dpse, 2024-04-22:
The ceiling is indeed reset when entering ramp mode (as to not interfere with standard regulation). But it is probably smarter to reset the ceiling whenever switching modes and then just step down again. [Here](https://github.com/ToyKeeper/anduril/files/15061134/hex.zip) are some updated hex-files that do just that.

Carpe-Diem-Postero, 2024-04-26:
Hi, I just came back from a work congress. Sorry for not being able to properly test this, I swear I'm on it, it's just that had a couple of  troubled weeks. Tomorrow I have full 24h work shift, but next week I should have time to begin tests.

Thanks for your hard work, dpse. I hope I can contribute ASAP.

dpse, 2024-04-27:
It's no rush! I'm just glad to get some feedback. I've made some [further changes](https://github.com/ToyKeeper/anduril/files/15137614/hex.zip) in case you want to test: The ceiling is now adjusted gradually (i.e. over time) so that the change should not be visible and the bike flasher level is also automatically adjusted to not clip bursts/flashes.

Carpe-Diem-Postero, 2024-04-27:
> It's no rush! I'm just glad to get some feedback. I've made some [further changes](https://github.com/ToyKeeper/anduril/files/15137614/hex.zip) in case you want to test: The ceiling is now adjusted [...]

OK, so far I have tested this latest version in two Wurkkos TS10 V2 (RGB), One TS10 V1 (Single aux) and Wurkkos FC11 (Sofirn sp10 pro refuses to overheat, due efficient heat dissipation, so I'm not counting this one). I have tested Candle mode, Strobe and Bike, ramping brightness to full, and measuring lux over time.

Candle mode seems to work perfectly fine, as does Strobe. Both keep brigtness up until the light starts to overheat, then it steps down for a controlled brightness. If I them cool down the light using state-of-the-art techniques like holding it in my hands, then ceiling goes up and the light gains power again.

So, Candle and Strobe works fine.

Bike, on the other hand, steps down dramatically (the FC13 goes from 112000 luxes to 400. Then it stays at that ceiling even if the light cools down. While in that "step down" mode, holding the button produces "turbo", raising brightness to max, but as soon as you release the button, it goes back to 400 luxes.

It seems that bike mode is not recovering from the step down state.

I will report behavior in Wurkkos TS11 and Noctigon KR4 ASAP.
Cheers.

dpse, 2024-04-27:
Thanks! I took a look at bike mode again and I had messed up the brightness scaling. [Here](https://github.com/ToyKeeper/anduril/files/15139564/hex.zip) is a new attempt at fixing it. I've not had any time to test it yet though...

Carpe-Diem-Postero, 2024-04-27:
> Thanks! I took a look at bike mode again and I had messed up the brightness scaling. [Here](https://github.com/ToyKeeper/anduril/files/15139564/hex.zip) is a new attempt at fixing it. I've not had any time to test it yet though...

Preliminar test with TS10 RGB and FC13 shows that bike mode step down is now much more progresive. FC13 went from 112000 to around 45000, and stayed at that level. Good work!

The only caveat seems that bike mode still does not recover from the step down, even if the temperature drops. Candle and strobe restore ceiling level when the temperature drops, but bike mode does not. the only way to reset the ceiling is to turn it off.

dpse, 2024-05-02:
I've tried to improve it a bit more [here](https://github.com/ToyKeeper/anduril/files/15192673/hex.zip). The tricky bit is that the flashes in bike mode are twice as bright (at least in terms of output level) as the steady level so the steady level needs to be scaled down way below the ceiling to prevent the flashes from being cut off. This makes it appear kind of dim even though the actual flashes are at really high level.

I put together some measurement equipment the other day so I can properly measure temperature and illuminance over time and tested bike mode today and everything seemed to work fine. Applying a bit of active cooling after stepping down I can see it slowly step up until the thermal ceiling is hit again.

Carpe-Diem-Postero, 2024-05-09:
Ok, i'm downloading this new version to check. Will report back soon.

Carpe-Diem-Postero, 2024-05-09:
Ok, i'm back with the report.

I tested latest hexes with TS10, TS10-RGB, TS11, FC13 and D4k.

Every single of them worked perfectly fine. Bike mode ceiling now slowly raises up when the light is cool down.

Amazing work!! 🙇🏻

dpse, 2024-05-13:
That's great!

Carpe-Diem-Postero, 2024-05-15:
Amazing, I saw you made a PR request with said changes.

It is safe to assume that this changes will be merged in the next release?

SammysHP, 2024-05-15:
> It is safe to assume that this changes will be merged in the next release?

That entirely depends on the maintainer of this project. ;) Currently there are 24 other pull requests not merged yet.