ToyKeeper/anduril #62: Police Strobe no output when main channel has lowest output saved

duncanyoyo1, 2024-04-11:
I can't get the police strobe blink mode to function correctly. It seems like there might be a glitch in the mode memory.

Using the latest trunk

If the light is set to moonlight ( 1/150 ) for the main channel, switching to the police strobe results in no output. 1H doesn't do anything in this mode.

Turning the light off, back on to main channel and setting it above 1/150, then going back to the strobe there's output as expected.

Seems like it's just an issue with the lowest mode.

I have this issue with the TS10-rgbaux, TS11 and TS25.

I thought I was doing something wrong for a few hours till I figured out it was the output setting that was causing the issue.

The firmware that shipped with my TS10 didn't have this issue, though I am not certain what version it was. I would have to flash it back and check it. The TS11 and TS25 both shipped with firmware versions that didn't have the police strobe.

Seems very strange as the police strobe doesn't seem to even have a brightness setting. So I have no idea why it being on moonlight for the main channel would result in no output on the aux.

SiteRelEnby, 2024-04-11:
At the moment level 1 for aux channels is off. See https://github.com/ToyKeeper/anduril/pull/47

duncanyoyo1, 2024-04-11:
Ah well that checks out then. Thank you.

Still seems like a bug as I thought the brightness wasn't an option for police strobe ( e.g. shouldn't it always be aux high? )

Trying to use that PRs changes I get a compiler error. I also tried to do the same thing myself ( though slightly different, not with the bitwise operations ) and am hitting the same error. I likely am just not familiar enough with the project, missing an include somewhere or something silly. Or it's probably just out of date with the main branch, but I'm not sure what all changed or where to start or look. Any help is always greatly appreciated.

```
In file included from hw/wurkkos/ts25/hwdef.c:7:0,
                 from ui/anduril/anduril.c:88:
./fsm/chan-rgbaux.c:6:1: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘void’
 void set_level_auxred(uint8_t level) {
 ^
```

aslotnick, 2024-04-18:
Police strobe uses the current memorized level, so that matches my expectation with the problem of level 1 for chan-aux and chan-rgb-aux resulting in "off" rather than "aux low".

https://github.com/ToyKeeper/anduril/blob/trunk/ui/anduril/strobe-modes.c#L240

```
inline void police_color_strobe_iter() {
    // one iteration of main loop()
    uint8_t del = 66;
    // TODO: make police strobe brightness configurable
    uint8_t bright = memorized_level;
```

I do think #47 would fix this issue. I probably should have opened an issue alongside my PR for clarity as well!

I'm unsure about the compiler error, but I do know that my version was able to compile within github using the workflow Toykeeper setup but in my own repository. If you go to the "checks" section of https://github.com/aslotnick/anduril/pull/1/checks you can download the resulting artifact ([direct link](https://github.com/aslotnick/anduril/suites/22880773213/artifacts/1424282439)] in case you care to try it out. I've rebased from the latest version of `trunk`.

closed 2024-04-19 by ToyKeeper

ToyKeeper, 2024-04-19:
Fixed by 5a69f495bf773d825468ab8ce74b79b3d3de0dc5.

Police strobe uses the same adjustable brightness method as beacon mode, because on some lights (dual channel lights), police strobe is on main LEDs instead of aux LEDs.  But it could definitely use a built-in brightness adjustment, and perhaps disable the variable brightness entirely when configured for aux LED channels.

PR #47 (2024-04-19): Bugfix: aux channel modes should use low brightness for level 1 (not off)
In the current implementation of using aux LEDs as channels, level 1 (the lowest level) results in the aux being set to "off" while higher levels result in the aux set to "high".

With this change, level 1 will result in aux "low" while higher levels continue to result in aux "high".

This makes the chan-aux and chan-rgbaux feature slightly more useful. Fixes #65 , also fixes #62.

## Testing
* ts10 v1 (single aux): low aux works for the aux channel at level 1
* ts10 v2 (rgb aux): low aux works for all channels (R, R+G,  [... cut]