ToyKeeper/anduril #65: When using aux LEDs as channels, the lowest mode turns the LEDs off

aslotnick, 2024-04-18:
This applies when using a light with aux LEDs such as the ts10 or ts10-rgb.

Steps to reproduce:
* after factory reset, enter advanced mode. On ts10, smooth ramp floor level will be 1/150.
* using 9H from on, activate an aux LED as a channel.
* Using 3C from on, select the now active aux as the active channel.
* Select the floor level. The aux LEDs will be off.

My expectation is that when using an aux LED as an active channel, the light being on at level 1 should result in the aux LEDs being active, but this doesn't seem to be the case. When ramping to any other level (2+), the aux LEDs are on at the high level.

SiteRelEnby, 2024-04-18:
https://github.com/ToyKeeper/anduril/pull/47

ToyKeeper, 2024-04-19:
Fixed by 5a69f495bf773d825468ab8ce74b79b3d3de0dc5.

PR #47 (2024-04-19): Bugfix: aux channel modes should use low brightness for level 1 (not off)
In the current implementation of using aux LEDs as channels, level 1 (the lowest level) results in the aux being set to "off" while higher levels result in the aux set to "high".

With this change, level 1 will result in aux "low" while higher levels continue to result in aux "high".

This makes the chan-aux and chan-rgbaux feature slightly more useful. Fixes #65 , also fixes #62.

## Testing
* ts10 v1 (single aux): low aux works for the aux channel at level 1
* ts10 v2 (rgb aux): low aux works for all channels (R, R+G,  [... cut]
closed 2024-04-19 by ToyKeeper