ToyKeeper/anduril #75: Request: Don't use Smooth Steps for Momentary Moon/Momentary Low (from lockout mode)

lukestein, 2024-05-26:
I love smooth steps, but feel they feel a bit off when just trying to get a bit of momentary light.

I note that momentary mode and tactical mode don't use smooth steps. Would politely suggest/request that the 1H/2H from lockout momentary moon/low also turn on and off directly without using smooth steps.

SiteRelEnby, 2024-05-26:
I set a moderately high level as my 2H (manual memory) on my lights, but I agree it's probably unnecessary for moon level, but potentially would be missed for higher levels by some users.

Thoughts on a few options:
* Don't use smooth steps for momentary moon (ever), use if enabled for momentary low
* Make the configuration for smooth steps something like off/exclude momentary lockout modes/always
* Make a separate config item somewhere - at the moment there's a PR for optional smooth steps while in tactical mode (https://github.com/ToyKeeper/anduril/pull/21) as an example.

lukestein, 2024-05-27:
Thanks for thinking deeply about my off-the-cuff suggestion @SiteRelEnby

I have no strong feelings although realize we already have a lot of config options. Obviously people use these momentary-from-lockout modes differently so a key thing would be accommodating this diversity.

Simplest good solution from my perspective might be your option one: No smooth steps for moonlight from lock