ToyKeeper/anduril #77: D3AA has a bug when manually exiting version check.

maru801, 2024-05-29:
I noticed that if you do 15C to get into the version check mode with the D3AA, when you manually exit by doing a single click (to skip the rest of the version check), you see the LEDs quickly buzz/flash with a dim light for a second or two. If you only have the main LEDs flash the numbers (version check, battery check, temp check), the main LEDs will flash only. However, if you have any AUX LEDs flash the sequence, the buzz/flashing will be both the AUX and main LEDs together.

I compiled the current github code for the D3AA hex's file, so this bug is on the latest release (v2024-04-20).

SiteRelEnby, 2024-05-29:
Can also reproduce this (not just on the D3AA). Looking quickly at the code, my guess is version_check_state should maybe have its own 1C handler in anduril.c that catches it before `version_check_iter()` is called again, and when 1C does get caught it's already called it (multitasking has weird edge cases in some areas). Will try to do a test build later when I have the energy to.

maru801, 2024-06-09:
Found out that this also affects the 3C from off battery check, temp check, etc.

If you enter to battery check, manually exiting will have this same behavior. Furthermore, switching check modes with 2C will also very briefly show this behavior. One difference though is that of the three D3AA's that I have, they vary slightly in the amount of time that the aux LEDs buzz before stopping.