ToyKeeper/anduril #79: Request: Make tactical mode slots stay on after holding the button for X amounts of seconds.

maru801, 2024-06-02:
For those that will ever use tactical mode, It would be nice to have a feature allows you to click to the desired tactical slot and have it stay on after you hold for X amount of seconds. I would imagine for any situation that people would actually want tactical mode for, not having to keep the button pressed down 100% of the time would free up more parts of their hand to be used elsewhere. Then the user simply clicks once to turn the current tactical slot off.

I don't know how many seconds would be appropriate, or if a user selected value would be preferred here, but it seems like a nice idea for tactical mode.

SiteRelEnby, 2024-06-02:
I like this idea. Perhaps an option in the tacmode config menu for a number of seconds (0C = always use momentary) before it locks the mode on?

maru801, 2024-06-03:
That sounds great. The config menu for this in tac mode could be set to a 4th flash when you 7H to get to its config settings (where you then click X amount of times to set the seconds to wait before locking the state).

fiveseven808, 2024-10-01:
I like this idea! It would be great to have the light stay on after 10 seconds of holding or something, especially tactical level 2! User selectable value would be best though. 

It's my understanding that thermal limits are ignored during tactical mode though?

dirtydancing, 2025-02-09:
> It's my understanding that thermal limits are ignored during tactical mode though?

Correct, currently there is no thermal regulation in tactical mode, cf. https://github.com/ToyKeeper/anduril/issues/121.