ToyKeeper/anduril #81: D3AA bug: Sometimes can't dim from top of ramp

lukestein, 2024-06-07:
*Flashlight*: Emisar D3AA, 519a
*Version check*: 0161-2024-04-20

---

I am getting an occasional issue where my D3AA get stuck at the top of its ramp and can't dim from there with 1H or 2H; it can only  turn off (1C).

If I *do* turn off from this state, it will get stuck again if it hits top of the ramp. Unscrewing the head sometimes fixes this, but doesn't always. I note the issue seems to arise after screwing on the head yields a quick triple flash.

This issue came up a few weeks ago. A factory reset seemed to get things sorted out, but the issue is now (occasionally) back. I've been waiting to try to get a video and finally did:

https://github.com/ToyKeeper/anduril/assets/44452336/d18e34e2-27e2-411d-9479-85ce954ee60f

Setup is currently advanced mode, stepped ramp running from 1 to 91 with 7 steps. Happy to provide other details if I can and they'd be helpful.

Light-Veteran, 2024-06-26:
Have you try to reinstall the same firmware?

lukestein, 2024-06-26:
@Light-Veteran: Nope, but good idea. I'll try later today.

SiteRelEnby, 2024-07-02:
Seems like a bug with the weak battery protection to me. What battery are you using?

lukestein, 2024-07-11:
Thanks @SiteRelEnby

Agree it seems to be an issue with battery protection. I never seem to get this issue unless I'm below about 3.6v, and it's more frequent as voltage gets lower. I only have Wurkkos 14500 cells, but it's occurred with several different ones.

> Seems like a bug with the weak battery protection to me (probably something around `ramp_level_hard_limit`) since it hasn't been seen on any other lights.

https://github.com/user-attachments/assets/783f64be-3b2e-4b5d-99da-19265ebc32eb

SammysHP, 2024-07-12:
The three flashes at the beginning indicate that a weak battery was detected. The voltage check also shows a much lower voltage at first.

lukestein, 2024-10-11:
I'm wondering if this is the same issue: Weak battery detection for some reason disables the ramp? https://www.reddit.com/r/Hanklights/comments/1g0v7mo/d3aa_stuck_on_particular_brightness_level/

lukestein, 2025-02-19:
Looks like another new report:

https://www.reddit.com/r/flashlight/comments/1isey5m/anduril_2_d3aa_issue/

cc: @SammysHP

SiteRelEnby, 2025-03-03:
Got this one on my radar to look at. Trying to break out of a bit of depression recently and do something more productive than doomscrolling and posting, trying to work up some motivation to put some more time in on the project.

Light-Veteran, 2025-03-03:
> Got this one on my radar to look at. Trying to break out of a bit of depression recently and do something more productive than doomscrolling and posting, trying to work up some motivation to put some more time in on the project.

I'm sorry about your depression and i hope you wake up easily. Remember that your work in this and other communities is essential to making us all happy and probably you too.
I lost my mother in August and just doing a lot of things helped me to move forward.
A big hug