ToyKeeper/anduril #83: Flash in lockout mode with '1h' or '2h' and a low light setting

void-spark, 2024-07-07:
Hi!

I'm trying to move my Hank d4k tint ramping dual channel from old firmware
(model 0136 date 2021-12-13) / anduril.2021-12-13.emisar-d4sv2-tintramp-fet.bin
to the current firmware (still model 0136) / anduril.2024-04-20.hank-emisar-2ch-fet.hex

I learned a lot about the new channel modes and other new things.. :)

I tend to use lockout mode a lot, and have it set to a fairly low light level (Actually, ramps are still set to defaults).
When I turn on the light (in lockout, staying in lockout, so 1H or 2H), there is a flash just at the start.
I'm not sure what causes it, maybe the jump start value which seems to be a bit higher for this 2ch-fet version?
Or something related to the smooth steps feature?
I need to run some tests with those two, but I've been at it for much too long, need a break :)

I'm currently running this build, but the same was observed with the 'vanilla' build:
https://github.com/ToyKeeper/anduril/compare/trunk...void-spark:anduril:d4k_tint_ramp
This build mainly just makes blend mode the default and only enabled mode, makes things a bit simpler for a tint ramp :)

Settings currently used with this custom build (I wish these maps could be generated automatically in the build :D ):
a500 010a 0a8c 8c01 0700 0000 0308 0003
0300 0000 8000 0000 0000 0000 0000 0000
8000 0000 0000 0000 0001 0300 0000 0000
2943 3202 072d 0029 1304 058c 1e99 28

a5 start marker
00 ramp style (0 = smooth, 1 = stepped)
01 ramp 2C style
     0 = no turbo
     1 = A1 style: Off 2C = ceil, On 2C = turbo
     2 = A2 style: Off 2C = ceil, On 2C = ceil, Ramped ceil 2C = turbo
0a Ramp smooth floor
0a Ramp stepped floor
8c Ramp smooth ceil
8c Ramp stepped ceil
01 Ramp smooth speed
07 Ramp stepped steps
00 Ramp after moon disabled/enabled
00 manual memory level
00 manual memory timer
03 (02) channel mode (0 = 'ch1', 1 = 'ch2', 2 = 'both', 3 = 'blend', 4 = 'auto')
08 00 channel modes enabled (bitmask). default = 1f = '00001000'
03 manual memory channel mode (0 = 'ch1', 1 = 'ch2', 2 = 'both', 3 = 'blend', 4 = 'auto')
03 blink channel (0 = 'ch1', 1 = 'ch2', 2 = 'both', 3 = 'blend', 4 = 'auto')
00 channel mode arg 0 '_' for 'ch1'
00 channel mode arg 1 '_' for 'ch2'
00 channel mode arg 2 '_' for 'both'
80 channel mode arg 3 'middle CCT' for 'blend'
00 channel mode arg 4 'warm-to-cool' for 'auto'
00 channel mode arg aux 0
00 channel mode arg aux 1
00 channel mode arg aux 2
00 channel mode arg aux 3
00 channel mode arg aux 4
00 channel mode arg aux 5
00 channel mode arg aux 6
00 manual memory channel arg 0 '_' for 'ch1'
00 manual memory channel arg 1 '_' for 'ch2'
00 manual memory channel arg 2 '_' for 'both'
80 manual memory channel arg 3 'middle CCT' for 'blend'
00 manual memory channel arg 4 'warm-to-cool' for 'auto'
00 manual memory channel arg aux 0
00 manual memory channel arg aux 1
00 manual memory channel arg aux 2
00 manual memory channel arg aux 3
00 manual memory channel arg aux 4
00 manual memory channel arg aux 5
00 manual memory channel arg aux 6
01 smooth steps disabled/enabled
03 strobe type (0 = party, 1 = tactical, 2 = lightning, 3 = candle, 4 = bike flasher)
00 strobe channel party
00 strobe channel tactical
00 strobe channel lightning
00 strobe channel candle
00 strobe channel bike flasher
29 party strobe delay
43 tactical strobe delay
32 bike flasher brightness
02 beacon seconds
07 voltage correction
2d thermal ceiling (45C)
00 thermal calibration offset
29 rgb led off mode
13 rgb led lockout mode
00 post off voltage seconds
05 autolock time
8c tactical level 0 (140)
1e tactical level 1 (30)
99 tactical level 2 (153 = max + 3)
28 jump start level

void-spark, 2024-07-08:
> Sounds like jump start.

Yup, certainly seems this, increasing from the old 15 (21) to the new 28 (40) on the old firmware also gives a far more noticeable flash.
The other way round on the new firmware lessens the flash, it's still more noticeable there, but I expect that might be better when I sort out some other issues I still see (tint ramping not very fluent, and on low level one channel tends to be off) :)

void-spark, 2024-07-08:
Hmm, the less smooth ramping seems to be because the top changed, not sure why that is, gonna have to dig through comments a bit.
Old tops for 0136: https://github.com/ToyKeeper/anduril/blob/274b615fc35b0c6256fc647422e98f721a06e8ce/spaghetti-monster/anduril/cfg-emisar-d4sv2-tintramp.h#L37
(Which is included in the actual 0136 file at https://github.com/ToyKeeper/anduril/blob/274b615fc35b0c6256fc647422e98f721a06e8ce/spaghetti-monster/anduril/cfg-emisar-d4sv2-tintramp-fet.h#L2)

New tops for 0136: https://github.com/ToyKeeper/anduril/blob/72128b62aae51c5747c8cc21e5113e145c66b9c7/hw/hank/emisar-2ch/fet/anduril.h#L54

void-spark, 2024-07-12:
Closing this, as the question is answered at least :)

closed 2024-07-12 by void-spark

DurvalMenezes, 2024-07-12:
I see this same behavior (a pre-flash when pressing 1H or 2H from lock-out) on my D3AA. Other users have reported it too. 

Is this a hardware thing or a software thing? Is it somehow created or aggravated by newer Anduril versions? 

I ask because neither my TS10 nor my FC13, both running older (pre-GitHub) Anduril versions, show that behavior. Haven't had the time to test either of them with the newer Anduril versions -- and I can't test the D3AA with pre-Github versions because there aren't any ;-)

TIA for clarifications.

SammysHP, 2024-07-12:
It's not related to a specific release, but to the configuration for the flashlight model. Some drivers need jump start, others don't.

DurvalMenezes, 2024-07-12:
> It's not related to a specific release, but to the configuration for the flashlight model. Some drivers need jump start, others don't.

The D3AA reportedly doesn't need it, but I (and other users) see it all the same.

Could it possibly be a bug in Anduril?

DurvalMenezes, 2024-07-12:
> I see this same behavior (a pre-flash when pressing 1H or 2H from lock-out) on my D3AA. Other users have reported it too.

Just adding more information: I also see the exact same thing when turning the light on  (ie, 1C from off) at the lowest moonlight (brightness level of 1, not sure if or at what level of brightness it stops happening or being perceptible.

SammysHP, 2024-07-13:
Have you tried turning it off and setting the floor to level 1?

void-spark, 2024-07-13:
Jump start is a software setting, configurable under 9C menu, if it's enabled.
But in the current version of the source (check which version your light is running though!!) I don't see jump start enabled at all for d3aa. So you might have to look elsewhere :)

DurvalMenezes, 2024-07-13:
Thanks @SammysHP  for the tips.

> Have you tried turning it off

Yep, by adding this to the bottom of my `hw/hank/emisar-d3aa/anduril.h`:

```
+//*** This flashlight reportedly needs no jump starting, so make sure it's off
+#undef USE_JUMP_START
+#undef DEFAULT_JUMP_START_LEVEL
``` 

And making sure it is *really* off where it matters, by adding this test/kludge at the relevant point of fsm/ramping.c:

```
      #ifdef USE_JUMP_START
      // maybe "jump start" the engine, if it's prone to slow starts
      // (pulse the output high for a moment to wake up the power regulator)
      // (only do this when starting from off and going to a low level)
      // TODO: allow different jump start behavior per channel mode
      // FIXME: don't jump-start during factory reset
      //        (it seems to cause some eeprom issues on KR4
      //         when doing a click with a loose tailcap)
+     #error SHOULD NOT HAPPEN!!!
      if ((! actual_level)
              && level
              && (level < JUMP_START_LEVEL)) {
          set_level(JUMP_START_LEVEL);
          delay_4ms(JUMP_START_TIME/4);
      }
      #endif
```
Compiling it with `.make emisar-d3aa` shows no errors, so I'm reasonably sure it's indeed turned off -- and the resulting hex, when flashed to my D3AA, still shows exactly the same pre-flash behavior both at 1C from off and 1H and 2H from lockout.

>  and setting the floor to level 1?

Not sure what floor you mean, but in my config I keep the default settings RAMP_SMOOTH_FLOOR to 1 and RAMP_DISCRETE_FLOOR to 10. Were you referring to either of these, or something else? 

Additionally: 
> neither my TS10 nor my FC13, both running older (pre-GitHub) Anduril versions, show that behavior. Haven't had the time to test either of them with the newer Anduril versions

I just tested the newest 2024-04-20 on my TS10 and I confirm it shows *no* pre-flash in any of the reported situations.

SammysHP, 2024-07-13:
> I just tested the newest 2024-04-20 on my TS10 and I confirm it shows no pre-flash in any of the reported situations.

Because it is not enabled for these targets.

> Yep, by adding this to the bottom of my `hw/hank/emisar-d3aa/anduril.h` … still shows exactly the same pre-flash behavior

Then it is a hardware limitation.