ToyKeeper/anduril #84: Request:  New Model Number for Single Channel D4SV2 with RGB Switch & FFL351A Emitter/Driver

Gymbow2007, 2024-07-17:
Reference discussion on BLF starting at message 725 [here](https://budgetlightforum.com/t/anduril-2-feature-change-suggestions/218045/729).

I’d like to see a new model firmware created for a D4SV2 single channel light with RGB switch using a FFL351A emitters and the FFL 12A linear no FET driver. Currently, this light ships using model 0135 dual channel firmware, and as a result, is not capable of having Police strobe with the Red / Blue Aux lights.  I also have to manually disable channels 2-5 so that the light switches between smooth & stepped ramp via 3C.

Note, this light ships from Jackson Lee and he has indicated that I should not used model 0137 as that may damage the emitters due to exceeding max current specs.  Although this may currently be a unique light, I believe as the new FFL351A emitters gain more acceptance, this combination will become more mainstream. These are incredible emitters with the potential to exceed the 519A in popularity.

SiteRelEnby, 2024-07-17:
👀

SiteRelEnby, 2024-07-18:
https://siterelenby.net/lights/anduril.hank-emisar-2ch-joined-nofet.hex - I don't have the appropriate hardware to hand, can you test it?

Gymbow2007, 2024-07-19:
Discussions I had with [Jackson Lee here](https://new.reddit.com/r/Hanklights/comments/1dllxqm/which_firmware_for_single_channel_d4sv2_with/) said that I shouldn’t use 0137 because it would risk overdriving the emitters. I can’t say that I fully comprehended everything he said, so maybe you can better understand it and determine which model (0135 or 0137) is best to use as the basis for modification.  

I’ll wait to hear back from you before deciding if I should test out the modded firmware above. And Thank all for your support!