ToyKeeper/anduril #87: Question: For Lightning, is 1H to "Interrupt current flash or start new one" a feature or not?

dirtydancing, 2024-07-19:
In the UI reference table at the end of the [user manual](https://github.com/ToyKeeper/anduril/blob/72128b62aae51c5747c8cc21e5113e145c66b9c7/docs/anduril-manual.md?plain=1#L1058), 1H from Lightning is listed as a feature with the action "Interrupt current flash or start new one". On the other hand, this feature is not mentioned elsewhere in the user manual, including [this overview](https://github.com/ToyKeeper/anduril/blob/72128b62aae51c5747c8cc21e5113e145c66b9c7/docs/anduril-manual.md?plain=1#L522-L534) and this [description](https://github.com/ToyKeeper/anduril/blob/72128b62aae51c5747c8cc21e5113e145c66b9c7/docs/anduril-manual.md?plain=1#L569-L574).

When checking with Emisar D4K (boost driver), version check 0273-2024-04-20, and Emisar D2, version check 0135-2024-04-20, initially I had the impression that this 1H is working. But when revisiting this, it appeared not to work, as lightning strikes started and ended seemingly independent of 1H button presses. So I am wondering if my initial impression that this 1H is working was due to the randomly generated lightning flashes, which may have coincided with the respective 1H button presses.

Could anyone clarify this, and possibly point to where this is addressed in the source code?

dirtydancing, 2024-07-19:
Thank you @SammysHP! Given that there is no mentioning of 1H for Lightning in the source code, to me, this also looks like 1H for Lightning is non-existent, although I cannot speak to the iter function that you referred to.

It looks like 1H for Lightning has been introduced to the user manual with the UI reference table all the way back on [2020-08-07](https://bazaar.launchpad.net/~toykeeper/flashlight-firmware/multi-channel/revision/537/ToyKeeper/spaghetti-monster/anduril/anduril-manual.txt#ToyKeeper/spaghetti-monster/anduril/anduril-manual.txt). If this is an error in the user manual, then I figure this line in the UI reference table should be removed.

So far, I have not even been using this (possible) feature at any time. This only came up when I added it to this [diagram](https://github.com/dirtydancing/anduril-tikz-diagram):

[image]

The only other diagram/cheat sheet I am aware of that also includes this is the cheat sheet by [Twinbee](https://www.skytopia.com/project/articles/anduril2-cheatsheet.html):

[image]

So for proper documentation, it would be good to know the status of this (possible) feature.

ToyKeeper, 2024-08-04:
A completed button click or hold release causes all "nice" delays to get interrupted and collapsed to zero.  This causes the current animation to be cancelled and start a new iteration of the main loop.  Usually it's not noticeable, but in lightning mode it can be used to start a new animation on command.

There is no specific handler for it... it's just how "nice" delays work.  It's written that way so that, if you're in the middle of battery check mode or something, and click to turn the light off... it'll turn off immediately instead of waiting until the current readout is done.  As a side effect, if you hold the button during lightning mode, then release when the LEDs are dark, it starts a new randomized "lightning strike" animation.

It's not a bug, but it's not really a feature either.  It's just a happy side effect which can occasionally be nice to know about.  So I documented it in the reference table.

closed 2024-08-04 by ToyKeeper

dirtydancing, 2024-08-04:
Thank you for your clarification and explanation! I will keep this in the diagram accordingly as well.