ToyKeeper/anduril #88: Request: Add support for 3V batteries

travisce, 2024-07-30:
Flashlights such as the Sofirn SC21 Pro running Anduril could benefit from being able to run CR123A 3V batteries. An example use-case would be a power outage, where the SC21 could be used as a mini lantern with a long-term stored pack of CR123As.

CR123As are rated at 3V, but will drop to 2.5V under load, which will trip Anduril's low-voltage protection (LVP).

Would it be possible/feasible to add a toggle for LVP to support 3V batteries (CR123As, or even potentially coin-cell batteries like CR2016) for flashlights running Anduril? The idea would be to have LVP enabled by default, but allow the user to turn it off if they want to run CR123As.

SammysHP, 2024-07-31:
There are very few flashlights (with Anduril) that can operate on such low voltage (the SC21 Pro isn't one of them). At most they would glow a little bit due to the forward voltage of the LED.

LVP is a safety feature that should not be user defined during runtime. It is already possible to change it during compile time for these rare cases.

ToyKeeper, 2024-08-04:
If a manufacturer makes an Anduril light designed to use 3V cells, I'll add support for it.  However, it's tricky to support 3V and 3.6V in the same build, because their voltage ranges overlap.  It's difficult for the light to know what type of battery is being used, so it might guess wrong and then break LVP and damage the battery.  It's not like AA/14500 support, where the voltages have no overlap, thus making it easy to detect the cell type.

So I'm waiting until 3V Anduril lights actually exist.  The SC21 isn't designed for that and won't perform well at low voltages.

In the meantime, if you have a SC21 that you only use with 3V cells, I'd recommend building your own version of the firmware with LVP disabled.  Or maybe use the voltage calibration function to make it think the battery is more full than it really is.  The manual shows only +/- 0.3V of correction available, but you can add more than 0.3V by clicking more times.  That way, instead of activating LVP at 2.9V, you could set it to activate at 2.5V or even 2.4V or 2.3V.  But the battery check function will show values higher than reality, so you'll have to convert the values in your head.

closed 2024-08-04 by ToyKeeper