ToyKeeper/anduril #94: Can't access strobe mode on dual-channel Emisar D4V2

fwiffo, 2024-08-26:
With a build from head, I'm not able access strobe mode with 3H from off. I verified that I'm not in Simple mode, and to be doubly sure, I built and flashed a version with Simple UI totally disabled. I verified that the appropriate #defines are enabled to include all the strobe mode stuff.

I don't know if this is a model-specific issue, or something specific to dual channel lights, or if it's a problem generally.

As far as I can tell, commit https://github.com/ToyKeeper/anduril/commit/4f7f90df760f1bd215ce1b2ec671d62b332690c6 (the fix for #85) is not the cause, but I could be mistaken.

fwiffo, 2024-09-01:
OK, I was wrong, it is that commit causing the problem. I've got a fix and will send a pull request.

Benik3, 2024-09-17:
I can confirm this bug, when I tried to compile PL47G2 without simple UI.  
Also I can confirm that the fix is working :)