ToyKeeper/anduril #102: Build failed on Sofirn and Wurkkos lights?

fiveseven808, 2024-09-23:
Not sure how to fix this issue. I can build fw3a binaries but I can't build sofirn and ts10 at least? (I haven't tried any else)

I get this giant output and am unsure how to figure out what's the problem:
```
===== anduril 2024-04-20+6#1 : sofirn-sp36-t1616 =====
In file included from /usr/lib/avr/include/avr/eeprom.h:38:0,
                 from ./arch/mcu.h:8,
                 from ui/anduril/anduril.c:38:
/usr/lib/avr/include/avr/io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
 #    warning "device type not defined"
      ^
In file included from ./arch/mcu.h:8:0,
                 from ui/anduril/anduril.c:38:
/usr/lib/avr/include/avr/eeprom.h:41:3: warning: #warning "Device does not have EEPROM available." [-Wcpp]
 # warning "Device does not have EEPROM available."
   ^
In file included from /usr/lib/avr/include/avr/eeprom.h:38:0,
                 from ./arch/mcu.h:8,
                 from ui/anduril/anduril.c:38:
/usr/lib/avr/include/avr/io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
 #    warning "device type not defined"
      ^
In file included from ./arch/mcu.h:8:0,
                 from ui/anduril/anduril.c:38:
/usr/lib/avr/include/avr/eeprom.h:41:3: warning: #warning "Device does not have EEPROM available." [-Wcpp]
 # warning "Device does not have EEPROM available."
   ^
In file included from ./arch/mcu.h:23:0,
                 from ui/anduril/anduril.c:38:
./arch/attiny1616.h:27:22: error: ‘CLKCTRL_PDIV_2X_gc’ undeclared here (not in a function)
     clock_div_1 =   (CLKCTRL_PDIV_2X_gc  | CLKCTRL_PEN_bm), // 10 MHz
                      ^
./arch/attiny1616.h:27:44: error: ‘CLKCTRL_PEN_bm’ undeclared here (not in a function)
     clock_div_1 =   (CLKCTRL_PDIV_2X_gc  | CLKCTRL_PEN_bm), // 10 MHz
                                            ^
./arch/attiny1616.h:28:22: error: ‘CLKCTRL_PDIV_4X_gc’ undeclared here (not in a function)
     clock_div_2 =   (CLKCTRL_PDIV_4X_gc  | CLKCTRL_PEN_bm), // 5 MHz
                      ^
./arch/attiny1616.h:29:22: error: ‘CLKCTRL_PDIV_8X_gc’ undeclared here (not in a function)
     clock_div_4 =   (CLKCTRL_PDIV_8X_gc  | CLKCTRL_PEN_bm), // 2.5 MHz
                      ^
./arch/attiny1616.h:30:22: error: ‘CLKCTRL_PDIV_16X_gc’ undeclared here (not in a function)
     clock_div_8 =   (CLKCTRL_PDIV_16X_gc | CLKCTRL_PEN_bm), // 1.25 MHz
                      ^
./arch/attiny1616.h:31:22: error: ‘CLKCTRL_PDIV_32X_gc’ undeclared here (not in a function)
     clock_div_16 =  (CLKCTRL_PDIV_32X_gc | CLKCTRL_PEN_bm), // 625 kHz
                      ^
./arch/attiny1616.h:32:22: error: ‘CLKCTRL_PDIV_64X_gc’ undeclared here (not in a function)
     clock_div_32 =  (CLKCTRL_PDIV_64X_gc | CLKCTRL_PEN_bm), // 312 kHz, max without changing to the 32 kHz ULP
                      ^
In file included from /usr/lib/avr/include/avr/io.h:99:0,
                 from /usr/lib/avr/include/avr/eeprom.h:38,
                 from ./arch/mcu.h:8,
                 from ui/anduril/anduril.c:38:
hw/sofirn/blf-q8-t1616/hwdef.h: In function ‘hwdef_setup’:
hw/sofirn/blf-q8-t1616/hwdef.h:83:5: error: ‘CCP’ undeclared (first use in this function)
     _PROTECTED_WRITE( CLKCTRL.MCLKCTRLB,
     ^
hw/sofirn/blf-q8-t1616/hwdef.h:83:5: note: each undeclared identifier is reported only once for each function it appears in
In file included from /usr/lib/avr/include/avr/io.h:634:0,
                 from /usr/lib/avr/include/avr/eeprom.h:38,
                 from ./arch/mcu.h:8,
                 from ui/anduril/anduril.c:38:
hw/sofirn/blf-q8-t1616/hwdef.h:83:5: error: ‘CCP_IOREG_gc’ undeclared (first use in this function)
     _PROTECTED_WRITE( CLKCTRL.MCLKCTRLB,
     ^
In file included from /usr/lib/avr/include/avr/io.h:99:0,
                 from /usr/lib/avr/include/avr/eeprom.h:38,
                 from ./arch/mcu.h:8,
                 from ui/anduril/anduril.c:38:
[... 554 lines cut ...]
./fsm/ramping.h:38:14: note: in definition of macro ‘GRADUAL_ADJUST_STACKED’
     if (   ((PWM ==   0) && (TARGET == TOP))  \
              ^
hw/wurkkos/ts10/hwdef.c:55:34: note: in expansion of macro ‘CH1_PWM’
     GRADUAL_ADJUST_STACKED(pwm1, CH1_PWM, PWM_TOP_INIT);
                                  ^
In file included from ./arch/mcu.c:9:0,
                 from ./fsm/spaghetti-monster.h:43,
                 from ui/anduril/anduril.c:80:
./arch/attiny1616.c: In function ‘mcu_adc_result_temp’:
./arch/attiny1616.c:98:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
./arch/attiny1616.c: In function ‘mcu_adc_result_volts’:
./arch/attiny1616.c:103:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
./arch/attiny1616.c: In function ‘mcu_adc_lsb’:
./arch/attiny1616.c:163:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
ERROR: build failed
===== 0 builds succeeded, 1 failed =====
FAIL: sofirn-sp36-t1616
```

SammysHP, 2024-09-23:
The device family pack for the MCU and you might need a different version of avr-gcc and avr-libc. Can you tell us more about your setup? The recommended way right now is to use the Docker image provided by https://github.com/SiteRelEnby/anduril-buildenv-docker (I still work with my original image, but it offers less comfort for the user, so I recommend @SiteRelEnby version).

fiveseven808, 2024-09-23:
I am unfamiliar on how to get setup with docker, can you recommend a tutorial? 

I'm using WSL running Ubuntu 22.04.5 LTS. To get setup, I followed the instructions under "Compiling" in the readme file.

fiveseven808, 2024-09-23:
Annnd nevermind I am dumb... I never ran the `./make dfp `command? it seems? I set things up last year so I didn't realize I missed it?
 After running that command, I am able to compile for the other platforms. Apologies for the unnecessary issue.