ToyKeeper/anduril #115: Request: No aux LEDs in Tactical Mode

dirtydancing, 2024-10-15:
This has recently been brought up by jon_slider over at [BLF](https://budgetlightforum.com/t/solved-anduril-tactical-mode-bug-report-aux-stay-on/225812). I agree, and I am bringing this up here so it does not get lost.

No aux LEDs in Tactical Mode would be consistent with no aux LEDs in Momentary Mode. Furthermore, it would avoid confusion over whether Tactical Mode or Lockout Mode is set.

Of course, with no aux LEDs in Tactical Mode, documenting the current state (Tactical Mode uses the aux LED settings of Lockout Mode) would not be necessary anymore, cf. https://github.com/ToyKeeper/anduril/pull/31.

fiveseven808, 2025-02-09:
I'd like to make this toggleable in a menu somewhere (maybe another menu in tac->7H , maybe 4th blink somehow?)

Switching off aux LEDs is not desirable to me, but I can understand the use case. Ideally I'd like to have tactical mode have it's own set of aux options. I.e. locked, unlocked and tactical mode have all different aux settings, but 7C and 7H don't really work well with how tactical mode is currently setup

dirtydancing, 2025-02-09:
Toggleable in tac config sounds good to me (an additional set of aux options for tactical mode may be a bit much, but why not).

Regarding an additional toggle option in the tactical mode config menu, here is a link to https://github.com/ToyKeeper/anduril/issues/79, in which another additional option has been suggested for tactical mode, possibly leading to two additional options.