ToyKeeper/anduril #117: Probable copyright infringement of tolkien and dc comics

bartoszp1992, 2024-11-15:
I have doubts about the use of the "Batt check" logo in the chart - it violates the intellectual property rights of DC comics.

It's similar with the name of the Andúril project.

Isn't it possible to change it somehow? Of course, I can fork it, but it would be nice if no one would pick on the author of the project when it gains even more popularity.
[image]

SammysHP, 2024-11-15:
"Anduril" is not a problem, it's even the name of a big company (but this firmware came first). The use of the logo should also be fine in this context and type of use.

bartoszp1992, 2024-11-15:
ChatGPT have doubts to that, and also on attached screen from tolkien estate, it can be problem

DurvalMenezes, 2024-11-15:
> ChatGPT have doubts to that

Current AI engines, ChatGPT specially, are known to hallucinate (produce totally false answers, even as they're rethorically  convincing).

IANAL, but I think this project's use of the name "Anduril" falls squarely under the "Fair Use" exception to Copyright Law, at least in the US and other countries following this doctrine: https://en.m.wikipedia.org/wiki/Fair_use

bartoszp1992, 2024-11-15:
> Current AI engines, ChatGPT specially, are known to hallucinate (produce totally false answers, even as they're rethorically convincing).
>
> IANAL, but I think this project's use of the name "Anduril" [...]

Okay, thanks. Seems legit.

bartoszp1992, 2024-11-15:
> > and furst anduril UI commits are from 2022.
> 
> No.

Right, my mistake. I checked that only on github commits.

ToyKeeper, 2024-11-15:
Copyright doesn't even apply to the name.  That's the realm of trademarks, and trademark law says this sort of thing is fine.  For more details on how that works, look up the case of Nissan Motors vs Nissan Computer.

As for the battcheck image, it may be covered by fair use, but that's something for courts to decide on a case by case basis.  It likely wouldn't even go to court though, because it's not worth paying a lawyer.  If DC cares, all they have to do is let me know and I'll change it, whether it's covered by fair use or not.  They don't seem to have a history of pursuing this sort of thing though, and they don't really have a reason to care anyway, since it's basically free advertising for them.

Given that GitHub currently has 13.4k repositories matching a search for the character that logo is for, and I can't find any instances of them attempting to strike that sort of thing, and the character's copyright expires in a few years anyway... I'm pretty sure there's no reason to worry.

Closing because no action is required.