ToyKeeper/anduril #119: Question: what are the src dir(s) for hank d4v2 w/ boost, d4v2 2ch, etc.?

jhoblitt, 2024-11-17:
I received my first anduril flash light a couple of days ago from Hank along with a flashing dongle.  Additional Hank lights are on the way.  Hank's website doesn't list the source config / .hex file used for each flash.  Out of an abundance of caution, could someone tell me which .hex files / source dir to use for these models?

- D4V2 w/ boost driver
- D4v2 w/ tint ramping (2ch
- DM1.12

Thanks!

SammysHP, 2024-11-17:
The flashlight is clever enough to tell you which version should be used. Every is explained here:

https://github.com/ToyKeeper/anduril/blob/trunk/docs/which-hex-file.md

jhoblitt, 2024-11-17:
:facepalm: @SammysHP Thank you.

My D4v2 is reporting its model as 0273: https://github.com/ToyKeeper/anduril/blob/trunk/MODELS#L44C1-L44C47 -- I would have never guessed that.

SiteRelEnby, 2024-11-17:
@jhoblitt the D4v2 as it currently is could be said to technically be a D4v2.5. The original D4v2 used a different driver (with 7135s instead of a linear), then the KR4 introduced the linear, and the D4v2 got the linear a while after that. That's why most of the the other single channel linears without their own unique hwdef use kr4 as well, e.g. on the linear+FET DT8K.

closed 2024-11-18 by jhoblitt