ToyKeeper/anduril #122: feature request: piezo buzzer low voltage protection alert (rear bicycle light)

gnomon-, 2024-11-28:
I sometimes use an anduril light as a rear bike light ([1](https://photos.app.goo.gl/c2KXvWmCS4kTi6wP7), [2](https://photos.app.goo.gl/cMuE24rCqWu8tuqf9)).  The one feature that I miss in this use case is an audible alert for when the battery is about to run flat.  On longer rides this can happen and leave me unaware of riding without rear illumination for a while, which is a risk I'd prefer to avoid.  (Checking cell voltage before setting out is a good safety habit but sometimes I'm in a hurry to get out the door.)  It would be useful for this particular case if anduril included support for pinging a small piezo buzzer to alert the rider prior to running flat.

Is this a reasonable idea to consider, do you think?  Is it even theoretically feasible?  Do the MCUs often driven by anduril have enough free pins, is the board footprint and typical driver circuit for a buzzer small enough to even consider adding to a flashlight driver design?  I hope so, I'd love to see some driver and light designs specifically aimed at this use case.

SammysHP, 2024-11-28:
Wow, that's a very exotic request! But interesting idea nevertheless.

> Is this a reasonable idea to consider, do you think?

I don't want to judge this.

> Is it even theoretically feasible?

Absolutely.

> Do the MCUs often driven by anduril have enough free pins

Usually yes.

> is the board footprint and typical driver circuit for a buzzer small enough to even consider adding to a flashlight driver design?

Nope. Even the inductor used for switching drivers is often a problem. There are tiny SMD buzzers available. But the main problem is see is similar to wireless control (Bluetooth): The driver is embedded in a pretty solid metal tube. Not sure you would be able to hear much from the buzzer.

(Btw, another request was to add a vibration motor.)

SiteRelEnby, 2024-11-28:
If you can design a driver with one, it can definitely be implemented.