ToyKeeper/anduril #123: momentary turbo with 1H while light is on

tinojaeger79, 2024-12-01:
It would be great to get to turbo with 1H and back after releasing the button while the light is on. So you can choose a brightness level for your actual need for general use and you have easier access to turbo and back. No need to make two times a double click. So many turbo bursts in a short time are very easy to do.

It could be implemented with 5C from on to enter this mode with actual brightness (and keep 5C from off for actual momentary mode). Brightness change could be done with 2H and ramp direction change with 2H again because 1H is needed for momentary turbo. So the general behaviour of Anduril 2 is that what the user expects.

Light-Veteran, 2025-01-24:
You can do with 3H from on

tinojaeger79, 2025-01-24:
Yes, but this is not comfortable to do it often.