ToyKeeper/anduril #127: Support for Wurkkos TS10 Max, TS10 SG, HD10, ..

gretel, 2025-01-31:
Wonder where Wurkkos publishes their firmware code to not violate the license. In the meantime, should these models be compatible with another one? Regards

SammysHP, 2025-01-31:
It's the same driver and firmware as for the regular TS10. There's no way for Wurkkos to not comply with the license because they can't even compile the firmware.

gretel, 2025-02-02:
> There's no way for Wurkkos to not comply with the license because they can't even compile the firmware.

technically, they can. the TS10 max has different firmware than the other models received at the same time.

besides, can you point me to any source of wurkkos declaring use of this GPL licensed code?

SammysHP, 2025-02-02:
> the TS10 max has different firmware than the other models received at the same time.

Mine had 0713-2023-07-29, same as the SG.

I don't know where they mention the use of GPL licensed code, but I also don't know which kind of agreement they might have with @ToyKeeper.

gretel, 2025-02-02:
> I don't know where they mention the use of GPL licensed code, but I also don't know which kind of agreement they might have with [@ToyKeeper](https://github.com/ToyKeeper).

the GPL is not subject to any "personal agreements". the vendor does ship products with firmware pre-flashed. if a individual gets benefits from a vendor that's a issue of it's own.

SammysHP, 2025-02-02:
> the GPL is not subject to any "personal agreements".

As the copyright holder of this code (a proper CLA may be useful in this situation), @ToyKeeper can do whatever she wants. From dual licensing up to a special permission to use it without attribution.

I'm still not sure about your intentions with this issue. Sofirn and Wurkkos have a long history of collaboration with the community and Anduril developers.

gretel, 2025-02-02:
two points:

1. i was asking about GPL compliance in product sales, not seeking sponsorship arrangements.

2. question about community exclusion: what's the rationale for treating my inquiry differently from others?

regarding copyright: yes, individual contributors retain copyright over their specific contributions unless they've signed agreements stating otherwise.

a proper CLA would have been signed prior to accepting any contributions.

SammysHP, 2025-02-03:
Well, to answer your initial question: Wurkkos does not have to publish the code, because it is published here and they do not modify it. In addition, for the GPL it is sufficient to provide the code on request. Wurkkos is so nice to include a link to the source in the manual.

[image]

gretel, 2025-02-03:
thanks. though, this reference is very worrying:

```
As the copyright holder of this code (a proper CLA may be useful in this situation), @ToyKeeper can do whatever she wants. From dual licensing up to a special permission to use it without attribution.
```

"dual-licensing", "special permission", "without attribution" are each and all a violation of the GPL.

i am kindly asking for a proper clarification so this has not to be checked by a suitable authority.

SammysHP, 2025-02-03:
> "dual-licensing", "special permission", "without attribution" are each and all a violation of the GPL.

They aren't. GPL does not prohibit dual-licensing. So if @ToyKeeper decides to publish Anduril under a different license or not release the code at all, that's perfectly fine, if all contributors agree or have signed the CLA. In this repo there are 1029 commits from ToyKeeper and 39 commits. I'm sure that most of the other contributors will agree with a license change. If not, it will be easy to remove the few lines of code contributed by them. You are not a contributor btw.

gretel, 2025-02-03:
> You are not a contributor btw.

this exclusivity is perceived as toxic on my side. CLA have not been signed, so dual-use is not legal. there is no discussion required. closing for now.

closed 2025-02-03 by gretel

SammysHP, 2025-02-03:
> this exclusivity is perceived as toxic on my side

Sorry, that wasn't my intention.

This is a small project and depends a lot on good faith of the community. If someone contributes to the project, it is assumed that the contributor trusts the maintainer and gives them permission to use the code in whatever way is necessary to fulfill the goals of the project.

But this issue started with the question where you can find the source code used on flashlights from Wurkkos Sofirn. It is here in this repository. They don't compile it themselves. Some versions from other manufacturers are based on forks (for example Fireflylite). But all of them have a history of working together with the community and if you see an indication for a license violation, it isn't intentionally.

gretel, 2025-02-03:
have been contributing, small parts, but under the GPL. again, CLA was not signed. reacting like this on a question in my own interest is bizarre.

gretel, 2025-02-03:
actually, now that i look at it again, looks more like my [little work](https://github.com/ToyKeeper/anduril/pull/1) was taken, not merged, so officially i did not have to become a contributor.

SammysHP, 2025-02-03:
Sorry, just looked at the authors in the git repo and - as you noticed - you are missing.

Let's wait for @ToyKeeper to give an explanation and decision how to proceed. In any case, there wasn't a license violation by Sofirn/Wurkkos so far.

ToyKeeper, 2025-02-03:
Hi, Wurkkos has been pretty good about making sure to follow the license.  Terry (from Wurkkos) has always been very cooperative.  Whenever there have been Wurkkos models missing from the repository, it hasn't been caused by any failing on their part.  Instead, the main factor in delays has been my poor health lately.  I got injured pretty badly and haven't been able to keep up with everyone's contributions, or with new hardware models.  Some people in the community have helped out, but it has taken me a while to get their work merged in.

It looks like Wurkkos's recent product pages don't have the required text to point people at the source code though, so I'll have to bug Terry about re-adding that.  Whoever writes their marketing copy probably didn't know it's important, and took it out because it didn't sound very marketing-y.  But it should be pretty simple to add.  And it's not technically even required.  The required clauses can be published in the manual shipped with the light, as shown a few posts above, and that satisfies the terms of the license.

> > There's no way for Wurkkos to not comply with the license because they can't even compile the firmware.
>
> technically, they can.

Sure, anyone technically can compile the firmware.  But they haven't actually done it that I'm aware of, and don't seem to know how.

> I also don't know which kind of agreement they might have with @ToyKeeper.

> As the copyright holder of this code (a proper CLA may be useful in this situation), @ToyKeeper can do whatever she wants. From dual licensing up to a special permission to use it without attribution.

> "dual-licensing", "special permission", "without attribution" are each and all a violation of the GPL.

Dual licensing is not a violation of the GPL.  The copyright holder can do basically whatever they want.  It's common to publish the code under GPL and *also* grant exclusive licenses of the same project, like how MySQL grants commercial licenses even though MySQL is open-source.  It's common and can be an effective way to fund projects.

Wurkkos has a license to sell products with the Anduril code.  However, they do so under the terms of the GPLv3, just like everyone else.  There is no dual license at play here.  Early on in the Anduril project, I would sometimes offer companies a choice of GPL or a custom license, where a custom license would cost a lot more, but I don't really like that so I set the price really high and nobody took me up on the offer.  Now though, since there are other contributors and no CLA, I've been using that as an excuse to not even offer any other types of licensing.  It would probably still be possible to dual license, or to change the license, similar to what DuckStation recently did, but I haven't tried and don't intend to.

Long ago, I made a conscious choice not to use a CLA, because I think it's better for everyone in the long run to divide the copyright among many contributors and thus basically force everyone to operate under copyleft terms.

> looks more like my little work was taken, not merged

Yeah, sometimes if a patch needs changes, it's easier to rewrite than to branch, pull, fix, and merge.  In some projects, rewriting is even considered a best practice, in order to keep the commit history cleaner.  That was also literally the first pull request, so I hadn't worked out a process for merges yet.  However, git and GitHub are showing that pull as merged, not merely closed, and it says your branch was merged.  So I'm not sure why you don't show up in the contributors list.  Maybe it's a github bug, maybe I just clicked the wrong thing... not sure.

> what's the rationale for treating my inquiry differently from others?

It did not appear that your inquiry was treated any differently from others, at the time you asked.  But a question like that will definitely get an inquiry flagged as contentious.  Following up with unnecessary escalations reinforces that impression.  Like changing the title to "Possible GPL Violation" *after* a photo of the manual was posted showing that Wurkkos are following the license.  Please don't do stuff like that.

ToyKeeper, 2025-02-04:
> i'm certainly going to "do stuff like that" when my personal rights are treated as above.

Maybe this will make it more clear why the response wasn't as positive as you may have hoped:

Let's say, instead of trying to find source code, the issue was about trying to find a family member.  The support ticket went something like this:

<**OP**> I wonder where my daughter is.  Haven't seen her in a while.  Anyone seen her?
<**Volunteer**> Have you checked her bedroom?
<**OP**> I think something may have happened to her.  Can you point me to where she said she's in her room?
<**Volunteer**> No, that's just kind of a normal place for people to be, so checking there is a good first step.  Like, that's where mine is right now.
<**OP**> I just want to know where my daughter is.  Why are you treating me so differently than everyone else?
<**Volunteer**> Okay, look.  Here's a screenshot of a selfie she posted like 20 minutes ago.  Looks like she's in her bedroom painting her nails.
<**OP**> I'm really worried.  I think I might need to get the authorities involved.  I'm changing the support ticket title to "Possible Kidnapping".
<**Volunteer**> Calm down.  Nothing happened.
<**OP**> This is serious.  Kidnapping is illegal.
<**Volunteer**> *sigh* I'll ping the person in charge.
<**Admin**> Here's a detailed answer, explaining why and how everything is fine ... but WTF.  Don't be like that.
<**OP**> Lol, I'm just asking questions.  I'll be like that any time I want.  I have rights.

Not all questions are "just a question".  Like, the famous example of asking "So where did you hide the body?" about someone who is neither dead nor missing.  People are gonna look at you funny if you ask questions like that.

Anyway, it appears no GPL violation has occurred.  Nothing to see here.  Maybe I can get Terry to add a note to the product pages in his store, so the code link will be in two places instead of just one.

gretel, 2025-02-04:
thanks for the lecture! suggest you read the GPL first, cause you obviously do not understand it's implications. though, you decide to make fun of me.

https://trainingportal.linuxfoundation.org/learn/course/inclusive-strategies-for-open-source-lfc103/course-introduction/course-information - this should suit you well.

ToyKeeper, 2025-02-05:
This is a bug report.  But it seems there is no bug, and nothing to report.  Just a strange argument over what appears to be nothing.  So it's a little hard to take seriously.  Regardless, I gave a serious answer, and in return received a "laugh" emoji and denial... which generally doesn't indicate a serious attempt to act in good faith.  So my second response focused on personal behavior rather than technical or licensing issues with the project.

But a moment ago when I scrolled and my mouse happened to hover over your name, I noticed something.  You have your account configured to request the use of joke pronouns, like "that" instead of "him".  Typically this is a thing some people do to ridicule the concept of pronouns and their importance in society.  But since pronouns indicate how a person identifies and wants to be treated, and your chosen pronouns are a joking form of mockery, one can only assume you want to be treated as a joke?

In any case, it does not appear any license violations have occurred here.  I think you'll find GPLv3 sections [6b](https://github.com/ToyKeeper/anduril/blob/e43203f814c7040c49dbb7d4804e9f2008633b7b/LICENSE#L257) and [6d](https://github.com/ToyKeeper/anduril/blob/e43203f814c7040c49dbb7d4804e9f2008633b7b/LICENSE#L275) cover this situation.  The source code in question is published for free, right here in this repository.  Wurkkos included a [prominent notice](https://github.com/ToyKeeper/anduril/issues/127#issuecomment-2630021758) with the shipped product, giving people clear directions to that code.  This is sufficient to satisfy the license.

The product also goes a step further by self-reporting additional information a user requires to locate the relevant parts of the corresponding source within the repository and build or install new versions... and instructions for accessing that additional info were given in the [very first response](https://github.com/ToyKeeper/anduril/issues/127#issuecomment-2628038609) to this bug report.  Info about how to use that data is given in [the manual](https://github.com/ToyKeeper/anduril/blob/trunk/docs/anduril-manual.md) and in a [special FAQ](https://github.com/ToyKeeper/anduril/blob/trunk/docs/which-hex-file.md) linked from the project's front page in the first bullet point under "Flashing Firmware", because it's the most frequently-asked question by a wide margin.

On a side note, the LinuxFoundation link you gave is not about GPL compliance.  It is a training course offered as [part of LinuxFoundation's DEI efforts](https://www.linuxfoundation.org/blog/blog/addressing-diversity-equity-and-inclusion-in-2021-and-beyond) (Diversity, Equity, and Inclusion) to improve racial and gender diversity in open source communities through the use of things like inclusive language.  And on that note, I'd like to refer you to the [*exact same course*](https://trainingportal.linuxfoundation.org/learn/course/inclusive-strategies-for-open-source-lfc103/course-introduction/course-information), to better understand why people should not use terms of mockery in the pronoun fields on open-source development platforms.

Anyway, since we're apparently using the bug report title to passive-aggressively fight about, um, something... not even sure what... I'll change the title to reflect that.

gretel, 2025-02-05:
where does vendor publish source code concerning use of my copyrighted code?

gretel, 2025-02-05:
legal note: stop the harassment, this is illegal at my jurisdiction. you mentioned having life issues already, i am sorry to hear, but that's the line you have crossed. time to take my friendly advise seriously.

ToyKeeper, 2025-02-05:
> where does vendor publish source code concerning use of my copyrighted code?

As stated clearly in the license in the clauses I linked earlier, if you would bother to read it, the vendor does not need to publish the source code themselves.  The vendor only needs to provide clear directions next to the object code (i.e. with the shipped product) saying where to find the source code, and it can be a network server operated by a third party.  The license is thus fulfilled by Wurkkos linking to this github repo from the product's printed user manual.

> legal note: stop the harassment, this is illegal at my jurisdiction. ... time to take my friendly advise seriously.

You do not appear to have a solid grasp of what is or is not legal, or what the license does or does not require, but I will happily take the advice offered in the DEI training course you recommended.  To foster a diverse community, I will ban anyone who mocks concepts of diversity, like race or gender, or who otherwise acts in bad faith.  Or, to put it in shorter terms, I believe the appropriate word is...  [*\*plonk\**](https://en.everybodywiki.com/Plonk_(Usenet))