ToyKeeper/anduril #132: Request: True No Memory Mode in Andúril 2 (Disable Temporary Brightness Memory)

hoanguyenvu7, 2025-05-14:
Currently, when Manual Memory is disabled (via 10H → 3 blinks), Andúril 2 still temporarily remembers the last brightness level if the user adjusted it before turning the light off.

This is confusing because users expect the "No Memory" mode to always turn the light back on at the ceiling level, regardless of prior manual dimming.

**Request:** Please make the current "3 blinks" (No Memory) option truly disable all forms of brightness memory, including temporary memory from manual adjustments.

**Expected behavior:**
- When Manual Memory is off (3 blinks),
- The light always turns on at the ceiling level,
- Even if the user dimmed it before turning it off.

No need for a new memory mode — just fix or clarify that Option 3 truly disables memory behavior completely.

This would make Andúril 2 truly predictable and perfect for users who rely on consistent startup brightness.

SammysHP, 2025-05-14:
> when Manual Memory is disabled (via 10H → 3 blinks)

What exactly is this option? Pressing 10H from on and releasing after the third flash lets you configure the "ramp up from moon" option.

From the manual:

---

  - `10C`: Activate manual memory and save the current brightness.
  - `10H`: Ramp extras config menu.
    - Item 1: Disable manual memory and go back to automatic memory.  
              (doesn't matter what value the user enters at the prompt)
    - Item 2: Configure the manual memory timer.  
              Sets the timer to N minutes, where N is the number of
              clicks.  A value of 0 (no clicks) turns the timer off.
    - Item 3: Configure whether to ramp up after `Off -> 1H`.  
              0: Ramp up after moon.  
              1: Don't ramp up, just stay at the floor level.
    - Item 4: Configure Advanced UI's turbo style:  
              0: No turbo, only ceiling.  
              1: Anduril 1 style.  `Ramp -> 2C` goes to full power.  
              2: Anduril 2 style.  `Ramp -> 2C` goes to ceiling,
                 or goes to full power if user ramped up to ceiling first.
              This value also affects momentary turbo in Ramp and Off modes.
    - Item 5: Configure "smooth steps".  
              0: Disable smooth steps.  
              1: Enable smooth steps.

Memory determines which brightness level the light goes to with 1 click
from off.  There are three types of brightness memory to choose from:

  - Automatic:  Always uses the last-ramped brightness.
    (does not memorize levels accessed by a shortcut,
    like turbo, `2C` for ceiling, or `1H-from-off` for floor)

  - Manual:  Always uses the user's saved brightness.

  - Hybrid:  Uses the automatic memory brightness if the light was only
    off for a short time, or resets to the manual memory level if it was
    off for a longer time.
    The timer for this is configurable from 0 to ~140 minutes.

Another way to think of it is:  There are three styles of memory for the
last-ramped brightness level...

  - Always remember         (automatic)
  - Remember for N minutes  (hybrid)
  - Never remember          (manual)

To choose a memory style, set the configuration accordingly:

| mem type    | manual mem  | manual mem timer
| --------    | ----------  | ----------------
| automatic   | off         | any
| manual      | on          | zero
| hybrid      | on          | non-zero

IAmJerv, 2025-07-14:
> This is confusing because users expect the "No Memory" mode to always turn the light back on at the ceiling level, regardless of prior manual dimming.

Anduril does not have a "No Memory" mode; only Manual, Automatic (which you seem to hate), and Hybrid where it forgets after a few minutes. **Manual Memory in Anduril is what "No Mode Memory" is in non-Anduril lights.** The behavior you desire would require setting the light to the ceiling then turning Manual Memory **ON** with 10C. The end result is that no matter whether you dimmed the light, the instant you turn the light off, it will forget where it was, and the next time you turn it on, it will come on at the ceiling. 

Option 3 has **nothing** to do with memory. All it does is determine whether or not you have to let off the button before it ramps up from the floor.

ToyKeeper, 2025-07-15:
As indicated in the comments, Anduril already has a "always turn on at the ceiling level" option:  Go to the advanced UI, ramp up to the ceiling, then click 10 times.  That tells it to always start at that level.

It sounds like there may be some confusion about what the other config options are, and things may have been changed by accident, so it might be a good idea to do a factory reset first.  For that, either do a 13H and keep holding for about 5 seconds, or loosen the tailcap, hold the button, tighten the tailcap, and keep holding for ~5 seconds.  The first method works on almost every light, while the second method requires a light with a side switch and an anodized tailcap.