ToyKeeper/anduril #134: Request: Ramp shape configuration in menu

Azirine, 2025-06-30:
level_calc.py defines multiple ramp shapes (ninth, seventh, fifth, cube, square, log) but this setting is not accessible in the menu. We already have 7H for ramp config menu, it would be useful to add ramp shape as item 4.

SammysHP, 2025-06-30:
The levels are pre-calculated and then hardcoded in the configuration file. That's the purpose of level_calc.py. Often there are also manual adjustments on top of the calculated values.

You want online calculation of the ramp, which is not possible with the current hardware (not enough memory).

Wildtimo, 2025-06-30:
Some mcu's have the space for a few extra pwm tables but I'm not sure if you can swap it out on the fly.