ToyKeeper/anduril #138: auto sunset timer option

ToyKeeper, 2025-08-13:
It'd be handy sometimes if the sunset timer would activate automatically when the light gets turned on.  Then you could have a light which takes just one click to turn it on at a memorized level, and it runs for a memorized amount of time then shuts itself off.  For example, a bedside light or something which sits in a dark hallway... it'd be nice to click once then try to sleep, or click once and walk through the dark area, and not have to worry about shutting it off or doing extra clicks to activate the timer.

Requested [here](https://budgetlightforum.com/t/anduril-2/62656/2042) but I've also thought about adding this a few times.  Filing a bug report to help me remember.

I'm not sure how the interface for this would work.  It'd need a number entry thingy added to a menu somewhere, to enter a value from 0 to 255 for the default sunset timer value, but I'm not sure where best to put it.  Or it could have something added to the sunset activation event, perhaps, to enable/disable the automatic timer in a more convenient way.

If I add a bunch of user-selectable UIs to Anduril 3, I'd kind of like to move Tactical Mode to its own dedicated UI, and potentially also move Momentary Mode to its own UI or make it even harder to activate since hardly anyone uses it.  Then the Ramp 5C action could be used to enable or disable the automatic sunset timer.

Or perhaps there's some other way to do it which makes more sense.

badgerj, 2025-12-16:
I like this idea.

It took me a few tries to understand that things like the memorized candle timer is "for this session only", and if you go back, you need to reset this the next pass through... and the next one... and... etc.

I may work with @ToyKeeper to see what they say and could do this, but want to fix up a few other things first.

SiteRelEnby, 2026-02-15:
I'm interested as well, but what I think would be good as a broader sunset timer rework is to also have a toggle between being a "sunset" type timer, and it holding at a consistent level (or as high as it thermally can if relevant), then turning off without *any* rampdown (other than smooth steps) and I've implemented it myself hacking around but it'd be a good addition, I think.

K3LOE, 2026-08-25:
I'm not a programmer or a particularly enthusiastic light UI enthusiast, I Just thought I'd share my use case.

I have a narrow pantry in which we keep tupperware and jars and kitchen appliances and such. It's like a small closet and it is dark with no built in lighting. The main breaker panel for the house is across from the shelving, so I currently have a Harbor Freight Braun 18650 adjustable worklight mag-mounted to it. 

Honestly? It's pretty great and it was cheap. My first 18650 experience!

But because I need it on for a few minutes at a time emptying the dishwasher for instance, I sometimes forget to turn it off when I'm done in the pantry. Not a huge deal, I usually find it on within a half hour or whatever, but occasionally it has burnt itself out which just seems wasteful...trying to solar charge more of my cells!

The benefit of having a timer on the main channel is obvious. I can dedicate a magnet-mounted light to that location for the sole floody purpose of lighting up my shelves from a short distance (hello mule??). I set the timer for 10 minutes because that's plenty for 90% of my pantry use and I decrease my rate of accidental battery discharges to effectively zero. Hopefully turning on the light with automatic timer is just a single click to get to your last ramp level, default 1C behavior

If on the rare occasion I need the light on for longer, I overide the timer setting somehow. Thanks to all real contributors!

Linterna28, 2026-08-28:
Hello. Thank you for the great Anduril2.
I use the sunset timer every night.
I would suggest activating a timer with a final 5-minute dimming.

1) From ON ... 8H to set the timer to the current brightness level.
2) Move 5C to 12H and use 5C to activate the timer.
3) 5 minutes before turning off, switch to sunset mode with gradual dimming.

Please move 5C to 12H, as I sometimes activate it accidentally and I believe much people too and I use sunset mode every night.

Thank you for your work.

Linterna28, 2026-09-14:
Hi there,
I’ve been thinking about timers, and a crazy idea just occurred to me.
How about an alarm-clock timer?
Since Luneoceans drivers have such a low moonlight mode barely visible in a pitch-black room you could set the flashlight to go to desired level after, say, eight hours at your desired brightness level.
Just think about it: it would be the first flashlight in history with an alarm-clock function.
⏱️🔦

SammysHP, 2026-09-14:
Unfortunately the clock isn't stable enough for this purpose. Every light would need its own calibration and even then it depends on temperature and battery voltage.

K3LOE, 2026-09-14:
> Unfortunately the clock isn't stable enough for this purpose. Every light would need its own calibration and even then it depends on temperature and battery voltage. - @SammysHP 

Anduril sunset mode already has a timer function. Why couldn't the same "good enough" timing be applied in this feature? Why would Toykeeper have started this enhancement ticket if she didn't think it was possible? 

The point wouldn't be to have atomic clock accuracy, but to ensure that your flashlight doesn't stay on for a battery-drainingly long time.

K3LOE, 2026-09-14:
> Hi there, I’ve been thinking about timers, and a crazy idea just occurred to me. How about an alarm-clock timer? Since Luneoceans drivers have such a low moonlight mode barely visible in a pitch-black room you could set the flashlight to go to desired level after, say, eight hours at your desired brightness level. Just think about it: it would be the first flashlight in history with an alarm-clock function. ⏱️🔦 - @Linterna28 

That's a cool idea, but I think it should be a separate enhancement suggestion so it can be considered on its own; it wouldn't be part of this feature request.

SammysHP, 2026-09-14:
> Anduril sunset mode already has a timer function. Why couldn't the same "good enough" timing be applied in this feature? Why would Toykeeper have started this enhancement ticket if she didn't think it was possible?

With its purpose it doesn't really matter if it runs for five hours or six hours. But for your "alarm" there might be a difference between waking up at 6 AM or 7AM.

Linterna28, 2026-09-14:
> That's a cool idea, but I think it should be a separate enhancement suggestion so it can be considered on its own; it wouldn't be part of this feature request.

Yes, you are right. Perhaps needs a feature request if it is possible to implement.

Linterna28, 2026-09-14:
> Unfortunately the clock isn't stable enough for this purpose. Every light would need its own calibration and even then it depends on temperature and battery voltage.

The light is never totally off only very very low with Luneoceans drivers, I don't understand why not is possible a timer to up light level after amount of time.

ThomasAH, 2026-09-14:
Please discuss this different feature in a different (new) issue

Linterna28, 2026-09-14:
> Please discuss this different feature in a different (new) issue

Ok. 
I'm going to try to start a petition.

Linterna28, 2026-09-14:
> > I don't understand why not is possible a timer
> 
> Because the timer is not stable enough for long duration. It might have an error of 10%, so it wakes you up an hour too early or too late.

OK, but the purpouse of the idea is not a wake up exactly but a timer to amount of time to remember something or to light up the flashlight.

Linterna28, 2026-09-14:
> Please discuss this different feature in a different (new) issue

How to do this?

I have tried to open a pull request and response is:

Pull request creation failed. Validation failed: must be a collaborator

Linterna28, 2026-09-14:
Created new issue:

Timer to light up the flashlight.

https://github.com/ToyKeeper/anduril/issues/179