ToyKeeper/anduril #142: aux button LED flickers high/low in candle mode when the main LEDs reach a certain brightness

ThomasAH, 2025-09-13:
The aux button LED is configured by default (adjustable via voltage config menu) to light up at low or high brightness level depending on the main LED ramp level.
This usually is what I want, but when using candle mode with a medium brightness the aux button LED sometimes flickers to high for a short moment each time the main LEDs reach a certain brightness.

I guess the aux button should use a constant value (e.g. currently set maximum or average brightness) to calculate low or high while candle mode is active. Only if candle mode is combined with a sunset timer, the aux button should switch from high to low once when the new value falls below the configured value for high button brightness.

Tested with r2025-07-07.sofirn-blf-lt1-t1616, but probably applies to other lights with aux side buttons as well.

badgerj, 2026-01-01:
May be related to this: https://github.com/ToyKeeper/anduril/issues/154

I'm very computer savvy and may be able to fix in software given some minimal direction. Even for a custom branch/firmware

SiteRelEnby, 2026-01-01:
Not a bug. The aux/button LEDs will switch from low to high at a specific threshold (iirc, it defaults at `DEFAULT_LEVEL`). I have a PR that was merged to the latest release that makes it configurable as to what threshold that happens at.

badgerj, 2026-01-03:
Thanks @SiteRelEnby . Where is this configureable?

I'd like to play with this and see if it is possible to change it for (LOW-HIGH) to possibly also include "OFF"?

As for some of us using a UV LED with a filter adding in additional light to is counter-productive. So we may just want to turn these off. See my linked report above. I am able to turn it off completely with a personal build, but I would like to make this configurable.

ThomasAH, 2026-01-03:
> The flickering is explicitly expected and desired in several modes, like lightning and strobe. The pulses there are sometimes less than 1ms long, and it is meant to light up the button for the same duration.

Ah, I wouldn't have expected it to be useful for lightning and strobe, but I don't mind it there.

> It is also intended to flicker during the main ramp mode, like when it blinks off for a few milliseconds to indicate that a setting has been changed or a ramp threshold has been passed. A dampening function would interfere with the intended behavior in most cases.

The dampening idea was just for the main-to-button case, not for general button light behavior.

But I understand that a general solution is a bit too complicated and I can live with the issue (or just adjust the threshold as a workaround), so feel free to close here, unless you want to keep it open as a backlog issue.

> Thanks [@SiteRelEnby](https://github.com/SiteRelEnby) . Where is this configureable?
> 
> I'd like to play with this and see if it is possible to change it for (LOW-HIGH) to possibly also include "OFF"?

This can already be turned completely off: 7H from batt check, options 3 and 4, each followed by 0C
(available in 2025-07-07)

SammysHP, 2026-01-04:
You can find the manual here:

https://github.com/ToyKeeper/anduril/blob/trunk/docs/anduril-manual.md